jeelabs/embello #5: Go under the hood using nohup

JohnOH, 2015-02-10:
```
john@odroid1:test-lpc8xx-none$ nohup lpc8xx -t -s 192.168.1.251:4000 > RasPi.txt
nohup: ignoring input and redirecting stderr to stdout
john@odroid1:test-lpc8xx-none$ cat RasPi.txt
found: 8120 - LPC812: 16 KB flash, 4 KB RAM, TSSOP16
hwuid: 1F300611C9981AAE528BDF51061900F5
entering terminal mode, press <ESC> to quit:

inappropriate ioctl for device
```

jcw, 2015-02-10:
Ah, this one caught me off-guard: you can't do that. To enter terminal mode, lpc8xx needs to have both stdin and stdout connected to the terminal (so you can type and see the output).

It should work without the "-s" though.

JohnOH, 2015-02-11:
Without the -s probably wouldn't need storing in  file. Perhaps a -l to log output or assume -l if no stdin match to stdout.  This topic woke me this morning the power of this gets scary when one considers Jeeboot. Perhaps one more abstraction layer to node name checking hwid.

jcw, 2015-02-11:
I suggest logging one level higher, i.e. in the terminal app (Putty?) you use to connect to Linux.

Also, look at the "telnet" and "nc" command: they too can connect to ser2net, and I'm pretty sure they can be used for logging. IOW, don't enable -s, but quit and then launch telnet/nc for capturing the data.

I don't want to replicate this functionality in what is essentially purely an uploader.

closed 2015-02-13 by jcw