jeelabs/embello #11: Enhancement - RTS/DTR toggling during -s serial session

JohnOH, 2015-02-24:
Add support for toggling the RTS and the DTR signals passed to the FDTI card either directly of via TCP. Perhaps <ctl>r and <ctl>d could be used. This would offer the ability to halt the running code by placing the LPC into reset.

JohnOH, 2015-03-09:
The ability to connect to the LPC without triggering a reset would also be very useful.

jcw, 2015-03-27:
Could be done similar to [PicoCom](http://linux.die.net/man/8/picocom) - ctrl-p pulses DTR and ctrl-t toggles it. Perhaps with ctrl-r to pulse RTS for a reset (no need for a toggle, I think).

JohnOH, 2015-03-27:
Toggle for RTS would I guess allow us to told the processor in reset - not executing code.

Best Regards, John.

jcw, 2015-03-27:
Ok, that suggests ctrl-o (off). Could indeed be of use.

closed 2016-12-05 by JohnOH