jeelabs/embello #17: Very low range.

LucasVinals, 2015-06-03:
Hi, I tried to pair both node and service (Arduino and RasPI) but somehow rf69spi.cpp in RasPi stopped working. I worked excelent for few minutes, but then, suddenly.. I't crashed. If I put both transreceivers close (< 5cm) they communicate, but as far I increase the distance, any message is received....

I tried every single output power, from 15, to 31, and frequency of pairing ("433" number is fine with my module of 433Mhz ???). 

Hope to hear from you..

Thanks!

jcw, 2015-06-03:
So rf69spi.cpp works briefly every time you start it up?

This might be crystal tolerances - different units can sometimes differ so much that the other unit has trouble catching on to them. Are both units using an RFM69?

Do both have an antenna, and did you double-check those connections? If you put them that close, then then only transmit level which will work probably, is "0".

LucasVinals, 2015-06-04:
Thanks for the quick answer!

No. It worked the first time I downloaded the code, follow the step and modified the variables according to my chip (node: 1 , network: 1, frequency: 433). Then, I turn off the Arduino (client sending request to my raspberry) and plug it again, but it didn't connect anymore.

Yesterday, I tried again and It worked! But today not. I'm really confused about what's happening here :S 
It has the same symptoms: if I put the two transreceivers almost together, it works. But as soon as I move them away far from 5cm, it stop receiving packages..

I'm wondering if it has to be with some module not loaded(ie. SPI module in kernel, some communication package..) or too much draining power (it's a RFM69HW. I know it's not recommended in the project page, but I put it with an independent power source).

I guess also if it could be an antenna problem, because it's about the range...  I dont know.

[Sorry for my english if I wrote some syntax mistakes]

Thanks again!

jcw, 2015-06-05:
Did you make sure that the ground of your power source and the RPi's are tied together?

LucasVinals, 2015-06-05:
Yes. They're tied properly. If not, I wouldn't have any connection at all.. I think it's a software issue, 'cause I did not modify any part of the circuit between those days.. One day was running fine, and then it stopped working when I move away the transmitter. RSSI varies from 170 ~ 160 (behind it stops).

LucasVinals, 2015-06-05:
Also, my power source can give ~130mA to RFM69 without issues (I tested the actual current drain from the module)

jcw, 2017-01-05:
This is an old issue - I'm closing it, but feel free to re-open if it's still relevant.

closed 2017-01-05 by jcw