jeelabs/embello #20: Vcc estimation can deliver unbelievable values

JohnOH, 2015-06-20:
Strange results when running [this](https://github.com/jeelabs/embello/blob/master/explore/1514-lpc/vcc/main.cpp) code on ESP-JNP. The code reliably reports Vcc as 27900mV this has been reproduced by @tve on similar hardware.

jcw, 2015-06-20:
If you look at the [code](https://github.com/jeelabs/embello/blob/master/explore/1514-lpc/vcc/main.cpp#L31), you can see that this means that `tap` must be 1. Which means `i` was 2, so the loop was exited on the very first iteration.

Maybe something is different right after upload? Is this same value still reported after you reset the unit without re-flashing it?

jcw, 2015-06-20:
Oh, wait - maybe it''s [this](https://github.com/jeelabs/embello/blob/master/explore/1514-lpc/vcc/main.cpp#L13) ?

JohnOH, 2015-06-20:
Good spot, I'll try it.

JohnOH, 2015-06-20:
Looking better at 3487mV

jcw, 2015-06-20:
Yuck. We may have to detect the chip at run time. Not hard, but yuck anyway...

JohnOH, 2015-06-20:
I had seen taht code but didn't put 2 & 2 together. Is equally odd if you reverse the comparator inputs?

jcw, 2015-06-20:
[Yes](https://github.com/jeelabs/embello/blob/master/lib/vendor/lpcopen/inc/acmp_8xx.h#L94-L99) and [yes](https://github.com/jeelabs/embello/blob/master/lib/vendor/lpcopen/inc/acmp_8xx.h#L111-L116).

jcw, 2015-06-20:
I've checked in a fix. Please re-open if this issue isn't resolved.

closed 2015-06-20 by jcw