jeelabs/embello #27: flib: PWM output does not switch off completely at 0

tht, 2016-12-26:
Looks like an output configured for PWM newer goes down to *exactly* 0.

Steps to reproduce:

````
120 PB6 +pwm  ok. \ pwm output 120 HZ
100 PB6 pwm  ok.  \ about 1% duty-cycle as expected
10 PB6 pwm  ok.   \ about 0.1% duty-cycle as expected
0 PB6 pwm  ok.    \ about 0.1% duty-cycle - NOT expected
````

This workaround fixes this issue by completely disabling pwm at 0.
````
\ Make sure pwm is completly off at 0
: pwm ( u pin -- )
  over 0 = IF
    -pwm  \ disable pwm
    drop  \ drop output value
  ELSE
    \ enable pwm output on pin again
    dup dup p2cmp 4 * bit swap p2tim timer-base $20 + bis!
    pwm   \ set pwm output (duty cycle)
  THEN
; 
````

There is no such issue on the upper end of the pwm output. `10000` sets the duty-cycle to exactly `100%`.

jcw, 2016-12-26:
Thanks for the report and workaround. I suspect that this is due to the way PWM is implemented in the hardware: it always toggles when the timer wraps around, and then toggles back when the preset counter value matches. So the best it can do at the minimum setting is to toggle very quickly, i.e. on and immediately off again - which produces an undesirable brief pulse.

tht, 2016-12-26:
Ahh… thanks for the clarification. So It's a limit on the hardware, not an issue in flib.

Most use-cases will never notice this 0.1% duty-cycle but it's an issue for me right now. I think we should add a note to the documentation and close this issue?

commit 65bb1b25 (2017-01-05): update pwm docs, closes #27
closed 2017-01-05 by jcw