jeelabs/embello #29: systick noch adjusted when running at 8MHz

tht, 2016-12-28:
Looks like the systick and `ms`, `millis` is not adjusted when running at 8MHz. Time is passing way too slow.

Steps to reproduce:

````
8MHz
millis .
\ wait about 1 second here
millis . \ about 100ms passed according to millis
1000 ms \ should wait 1s but waits about 100ms
````

jcw, 2016-12-28:
Ah - yes, a trap - after changing the clock speed, you need to call `1000 systick-hz` again to recalibrate the systick counter.

tht, 2016-12-29:
What do you think about recalculating the value inside 8MHz and 72MHz? This works for me:
````
: 8MHz ( -- )  \ set the main clock back to 8 MHz, keep baud rate at 115200
  0 RCC-CFGR !                    \ revert to HSI @ 8 MHz, no PLL
  $81 RCC-CR !                    \ turn off HSE and PLL, power-up value
  $18 FLASH-ACR !                 \ zero flash wait, enable half-cycle access
  clock-hz @ $E000E014 @ /        \ push systick-hz on stack for later use
  8000000 clock-hz !  115200 baud USART1-BRR !  \ fix console baud rate
  systick-hz                      \ consume systick-hz here
;
````
Same lines added in `72MHz` too. As it needs `systick-hz` this word has to be moved up in the file (or the frequency-changing words down…).

Looks like there is similar code in `clock.fs`. If we change `hal.fs` we should change all of them for all boards and also `hal.fs`.

tht, 2016-12-29:
Code written in comment above works perfectly fine but to be more exact `systick-hz?` should have a `1+` to compensate for `1-` in `systick-hz`. But if I change it to this the board completely locks up after a reset:

````
: systick-hz? ( -- u ) \ pushes current systick frequency on stack
  clock-hz @ $E000E014 @ 1+ / ;
````

Do you know why this happens? Something not completely initialized during power-up?

jcw, 2017-01-05:
Great idea to read back the clock (I don't know how the 1+ change could cause problems).

This is starting to affect a lot of places. The same adjust-after-clock-change also applies to baud rates. I wonder if we are going to need a more general mechanism - it feels a bit wrong to put this logic in the clock-changing code (some contexts won't use serial, for example - the UART peripheral might not even have been enabled).

jcw, 2017-01-15:
I don't think I'll adopt this suggestion - may in fact move the baudrate adjustment out of the 8MHz/72MHz/etc clock setting routines as well. There are many more clock changes one may need to use, and the USART1 serial port is not always in use (or even enabled).

Reading back the clock has been implemented as suggested, i.e. `systick-hz?` - thx.

closed 2017-01-15 by jcw