jeelabs/embello #31: Forth: ms delay and multitasking

hexagon5un, 2017-01-03:
The "ms" function from the HAL section is very handy, but it doesn't play well with multitasking.

I wrote a multitasking version: 

    \ delay, yields while ticks < ms value
    : ms ( n -- )  
      ticks @ + 
      begin pause dup ticks @ < until drop ;

The question is where to insert it into your code stream.  It needs to be defined before any functions that use it, of course. Redefining `ms` in the multitasking file makes some sense, but it's definitely a HAL thing...  I guess if you defined it this way in the HAL file, `pause` would resolve to a `nop` and you'd still have a blocking milliseconds anyway.

These "milliseconds" are only milliseconds if you've set `1000 systick-hz`, of course, but that's the same issue that your `ms` function has too.  For everything I do, a wall-clock millisecond is a great thing to have around.  I wouldn't change that. 

Anyway, not a bug, but more of a feature request.  Hardly worth a pull request.  Not tested rigorously yet, but works for me.

jcw, 2017-01-03:
Great - you beat me to it, but that's what I had in mind for an upcoming excursion into multi-tasking!

Since `pause` does start out as `nop` in the core, it should be fine to adjust the `ms` definition in `hal.fs`. I also want to investigate further how this could all be combined with low-power sleeping.

hexagon5un, 2017-01-03:
Been doing just that!  It looks like `sleep`, which is used in the Mecrisp multitask lib that you include, enters a wait for interrupt sleep, which means that it'll get woken up by the systick every ms. If this is all the chip is doing, it's not a horrible drain.  But for real low power, you'd want to sleep for longer.

I'm not at all sure how the `lowpower-task` approach in `multitask.fs` will work with the way I'm implementing `ms` though -- I'd bet that I'm accidently keeping a task "running" even though it's continually pausing, which will prevent the lowpower mode from ever kicking in.

So yeah.  More thought needed about how they work together?  

Oh, and I noticed that `ms` is both in the `hal` and `clock` files for the STM32F1 series.  Not sure what you're planning with the code architecture there, but my guess is that it could work in either.

jcw, 2017-01-03:
I have yet to fully dive in, but `lowpower` seems to work off the assumption that tasks are suspended most of the time. We could perhaps create some timer words which will do exactly that: suspend a task and set things up so it gets restarted after a number of milliseconds. Then `lowpower` could kick in, and even query how much time there is to the next wakeup to select a lower-power sleep mode.

Ah yes, `clock` was split off from `hal` to reduce the amount of code needed for the USB driver version (there's a minimal `hal` in the `suf/` dir), but it's not been cleaned up or factored out in the rest of the code yet.

tht, 2017-01-04:
I'm looking into the multitasker too and there are some *problems* IMHO:

* The multitasker has currently no way to `sleep` one task a specific amount of time.
* Redefining `ms` allows tasks to sleep but prevents the lowpower-task from going to sleep.
* There is no way to register a word to be called in the future (timer callback).

I really like the idea of a callback-style timer task as it's done on NodeMCU, JavaScript, … but I'm not sure if this will be the best solution for all use-cases. One possible solution could also be to add a new state *sleeping* into the current multitasker which could allow the low power-task to sleep the microcontroller. But this is probably quite complex to do.

IMO an easier solution should be a timer-task which replaces `lowpower&`. It will executes words in a given interval or only once in the future. This task knows when something should happen and could enter sleep-mode in between.

Btw… this is the `ms` I'm using now. It *should* work even if there is an overflow in `millis`:

````
: ms ( ms_delay -- )
  millis swap   ( start ms_delay )
  begin
    pause
    2dup millis rot
  - < until
  2drop
;
````

tht, 2017-01-04:
I may have just found an easy solution to combine `ms` and lowpower-sleeping. Basically it introduces a new *state* into multi.fs.

    -1 \ existing: Task running
    0  \ existing: Task stopped
     1 \ NEW: Task waiting for interrupt

The new state get's executed in round-robin as normal but it does not count if the `lowpower-task` is checking if it's the only one running. This works because my (modified) `lowpower-task` makes sure it passes the cpu to every other *running* task after every interrupt.

First tests look very promising. Changes inside `multi.fs` are minimal and it does what it should do. I've pushed my changes to the new branch `tht` (e409ae194a5f9fe8c9455e76d673641c29d41b35).

jcw, 2017-01-04:
Yes, but...

This favours creating tasks, which really is expensive in RAM space use. On the F103, we have 20 KB of RAM, but on the L053's we need to make do with 8 KB. I'm not convinced that we really need to keep stack state around between activations and would still much prefer a timer callback mechanism.

In Mecrisp 2.3.3, Matthias just added a [quotations](https://github.com/jeelabs/mecrisp-stellaris/blob/master/common/quotations.txt) demo, using a nice `[: ... ;]` notation, so setting up little callbacks can be quite convenient.

hexagon5un, 2017-01-04:
@tht: You're totally right about the rollover in `ms`.  That's a sloppy, and classic, mistake!  If one cares about the rightness of milliseconds, your implementation is better.

tht, 2017-01-04:
Sure… IMHO this is **not** a replacement for a timer callback mechanism. This is an extension for the existing multitasker which adds a new task-switching `ms` without breaking sleep capabilities.

This allows a timer task to be implemented without having to worry about sleeping the CPU. I assume this also simplifies the timer task as it is now possible to use `ms` inside. But following this route needs a running timer task *and* lowpower task.

So combining the two tasks (timer and lowpower) is probably the way to go to keep memory usage as low as possible in the future. But still… the change I proposed above does not use a single byte of RAM and makes the multitasker easier to use today and for some use-cases in the future, where a timer based approach does not work.

jcw, 2017-01-05:
I'm going to hold back on the multi-fs "-1" change, because I think it can also be done with timers (i.e. set up a timer and have it suspend and wake-up the task), but I'll change the `ms` to be multi-friendly.

jcw, 2017-01-05:
Btw - one change I made is that if the arg is ≤0, then no task switch can occur - reasoning being that with "0 ms" you'll probably want to get back control asap. And it's roll-over safe due to `- 0<` iso `<`.

commit 4085e005 (2017-01-05): make ms multi-tasker aware, closes #31
closed 2017-01-05 by jcw