jeelabs/embello #32: rf-send used but not defined

tve, 2017-01-05:
rf-send is used in the forth code in several places but it is not defined. I hit the one in flib/any/varint.fs. A quick fix might be to change it to rf69-send, but that ties it to the rfm69 radio, which isn't great either.

```
1608-forth> egrep -r rf-send
prc/d:: t69try ( n - ) 16 rf69-power  0 <# #s #> 0 rf-send ;
oxs/l:  v> 0 rf-send ;
oxs/d-other:: t69try ( n - ) 16 rf69-power  0 <# #s #> 0 rf-send ;
flib/any/varint.fs:  v> 0 rf-send ;
rnw/d:: t69try ( n - ) 16 rf69-power  0 <# #s #> 0 rf-send ;
```

tve, 2017-01-05:
Would it make sense to change `rf69-` back to `rf-` wholesale? That would allow whichever radio code is included to work, as opposed to having different names for the same words in the various radios. It's not like we expect to have multiple radios of different types on the same node, right? (And if we did something else would probably be needed anyway.)

jcw, 2017-01-05:
Yeah, bit awkward - did I miss a few renames? I guess so.

There might well be a need for two radios when setting up some sort of relay, but there's no problem defining one, using it, and then defining the other (and ignoring the "redefined" warning). I agree that a full rename to drop the 69 is probably nicer in 99% of the use cases.

Ok, I'll do it (also need to adjust the weblog post - no big deal).

jcw, 2017-01-05:
Should be fixed in all `flib/` files now.

closed 2017-01-05 by jcw