jeelabs/embello #38: incorrect us timing in loop

jcw, 2017-02-05:
from @SevenW - this code produces unexpected results on the L052 / JeeNode Zero:

    : tt millis 100000 0 do 10 us loop millis swap - . ; Redefine tt. ok.
    tt 13637 ok.

Same code returns 943 on F103.

jcw, 2017-02-05:
Perhaps related: sometimes the value returned is not increasing monotonically, so `micros ... micros swap - .` can print occasionally print out a negative value.

SevenW, 2017-02-11:
*/ is 6 times slower then * / !
timing tests performed in a 1 million loop:
```
66 us    micros \ the default function
0.73 us  micros \ my low-resolution (32us steps) but fast implementation

64 us    ticks @ 1000000 clock-hz @ */
10 us    ticks @ 1000000 * clock-hz @ /  \ and this is mostly the division
 9 us    ticks @ 1000000 * clock-hz @ u/mod nip

1.01 us  ticks @ clkfactor * 14 rshift  \ no division
``` 
The trick is to replace the division by a division of a power of two, a right shift.
Now I adapted the micros function accordingly in hal.fs
```
4.07 us  micros \ using: begin ticks @ $E000E018 @ over ticks @ <> while 2drop repeat
2.69 us  micros \ using: ticks @ $E000E018 @
```
See https://github.com/SevenW/embello/blob/forth-ook/explore/1608-forth/flib/stm32l0/hal.fs#L92-L98

jcw, 2017-02-11:
Woohoo - splendid! Looks like the M0+ has no good hardware support for division.

jcw, 2017-02-12:
BTW, difference between `*/` and `* /` is that the former works with a 64-bit intermediate result.

closed 2017-02-16 by jcw