jeelabs/embello #39: cornerstone not defined in dad/h

binduwavell, 2017-02-05:
Please document pre-requisites if something needs to be loaded before this `explore/1608-forth/dad/h`. Alternatively if there are no pre-reqs (as I had assumed) there is a bug because cornerstone is not defined and is referenced.

I found cornerstone defined in always.fs in a few other folders, but it was not clear to me how these could/should be adapted for my STM32F103C8T6 (64K flash, 20K RAM).

I noticed a comment on your forums that you are in the process of some re-organization so possibly this was just an oversight related to those efforts. In any case, thanks a ton for all the great resources on getting forth up and running on a blue pill and for the very helpful libraries!

binduwavell, 2017-02-05:
Also just curious, what does <<<hal-dad>>> do/mean?

jcw, 2017-02-05:
Yes, the `h` (hardware), `l` (library), and `d` (development) files are the old setup (to be loaded in that order, BTW). I've switched to `always.fs`, `board.fs`, `core.fs` as new main app structure - see [this article](http://jeelabs.org/2016/06/thoughts-about-app-structure/) on the weblog.

Just for the record: the `cornerstone` definition was removed from `hal.fs`, as included by `h`, and is now in `always.fs`. But there are many more changes, so the h/l/d files are likely to be broken in several places now.

I merely used `<<<...>>>` as visual indicator, to make it very clear that these are boundaries in the dictionary which can be sliced - no big deal really, feel free to pick other conventions.

For a starting point of what to load in Mecrisp Forth on an STM32F103, I suggest using the files in `g6s/` (generic 64-pin serial) - these are maintained and I use them all the time.

binduwavell, 2017-02-06:
Thanks much for the response and pointers.

Given my board has 64k of flash and always.fs in g6s seems to work with 256k, do I need to adjust anything there?

Having read the article the whole dictionary boundary thing makes perfect sense, thank you!.

I'm using picocom with msend (per one of your articles) to interact with the blue pill. I've only connected TX, RX, VDD (5V) and GND. My USB Serial TTY device does have other connectors for DTR and CTS, I'm not sure if I need to set them up? It also has a jumper for 3.3v or 5v. I assumed that the blue pill might like receiving the full 5v. Also if I leave the ST-Link v2 USB stick connected it seemed reasonable to have common ground and VDD... Not sure if any of that makes sense I'm not much of a hardware guy.

Most of the time I use Ctrl-A, Ctrl-S and send `./always.fs` without issue. Then I use the same technique for `./board.fs` and as `hal.fs` is being loaded I'm warned about redefining the `bit` word and then the communication stalls. Sometimes it gets past that and warns about redefining the `init` word and then stalls. If I Ctrl-C and try to send board.fs again sometimes I see lines of forth code being echoed on the screen slowly. Sometimes after using `reset` or `eraseflash` I can get `./board.fs` to load although often with tons of `fatal error: systemstack called from unexpected goroutine` error messages.

I don't have any hardware hooked up that would call for loading `./core.fs` yet. So at this point I attempt to run `50 ms` and the system hangs forever. Or at least until I reflash over ST-Link v2.

Any advice would be greatly appreciated!

binduwavell, 2017-02-06:
I should mention that I've been loading your `f103-mecrisp.bin` rather than the slightly newer official Mecrisp-Stellaris binary.

binduwavell, 2017-02-06:
He `hello` word is printing:

```
hello 64 KB <g6s> 303CBB0A ram/flash: 19360 34816 free  ok.
```

binduwavell, 2017-02-06:
After many many repetitions, it appears that `./always.fs` and `./board.fs` loaded cleanly and `50 ms` is running without bricking things. Any idea why this was so flakey and if there is anything I can do to make it more reliable?

jcw, 2017-02-06:
You have to reset after `board.fs` is loaded, so it runs its new `init` which actives the SysTick timer.

W.r.t. `always.fs` being for 256K - it'll work for smaller too. In fact, that comment is wrong - I'll adjust it.

Please consider opening a new issue if you run into anything else - I'm about to close this one. Also, in case of questions and suggestions, I would suggest using to the [forum](http://jeelabs.net/projects/cafe/boards/) - this is more to make sure I don't miss potential bugs.

commit a44d2190 (2017-02-06): add comment about flash page size, closes #39
closed 2017-02-06 by jcw