jeelabs/embello #46: rf69.fs: Unable to configure network group id

tht, 2017-02-13:
I've first set all the variables to the desired values and called `rf-init`:

    8683 rf.freq ! 42 rf.group ! rf-init

Looking at the output of `rf.` you see that the second sync byte (should be rf.group) is still set to it's default value `0x2a`:

```
     0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F 
00: -- 04 00 02 8A 05 C3 D9 13 00 41 00 02 92 F5 20
10: 24 90 09 1A 40 B0 7B 9B 08 42 42 40 80 06 1C 00
20: 00 00 00 00 FF 00 07 80 00 C4 00 40 00 05 88 2D
30: 2A 00 00 00 00 00 00 D0 42 00 00 00 8F 12 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00
50: 13 81 88 08 00 00 01 00 1B 09 55 80 70 33 CA 08 ok.
```

SevenW, 2017-02-14:
What if you change to `8683 rf.freq ! rf-init 42 rf.group ! `?

SevenW, 2017-02-14:
And the more obvious observation: 42 = 0x2A?

tht, 2017-02-14:
Oh, copy-paste bug… it was set to rf.group 6 while testing. I did update the issue. Thanks!

Your suggestion also does not work:

```
8683 rf.freq ! rf-init 6 rf.group !  ok.
rf. 
     0  1  2  3  4  5  6  7  8  9  A  B  C  D  E  F 
00: -- 10 00 02 8A 05 C3 D9 13 00 41 60 02 92 F5 20
10: 24 9F 09 1A 40 B0 7B 9B 10 42 42 40 80 06 5C F1
20: 02 F7 2B 00 C2 00 07 D8 00 C4 00 00 00 05 88 2D
30: 2A 00 00 00 00 00 00 D0 42 00 00 00 8F 12 00 00
40: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 00
50: 14 45 88 08 00 00 01 00 1B 09 55 80 70 33 CA 08 ok.
```

Also this is not how it should be done according to the documentation:

> Variables: These should only be changed before calling rf-init

SevenW, 2017-02-14:
No you are right. `6 rf.group` needs to be doen before rf-init. What you could try to test is stand-alone (and mind the - in the word: `6 rf-group rf.` Just to test.

jcw, 2017-03-01:
Odd, it works here. Could it be dependent on the mode the radio is in? There are have been some mentions that the RFM69 has shadow registers, which only get updated under specific conditions.

JohnOH, 2017-03-01:
I think I made comment on *shadow* registers. I believe that setting RSSI threshold for instance does not take effect until receive mode is entered. If the radio is in receive mode then setting the RSSI threshold does not take effect until the mode of the radio is changed. I can appreciate that this approach would also sensibly apply to the DIOMAPPING register and by inference many other registers. I picture the registers being applied to the receiver as a mode change occurs.

jcw, 2017-05-05:
I think this has been resolved - @tht?