jeelabs/embello #51: Unexpected result of i2c. (/any/i2c-bb.fs)

doewie, 2017-03-02:
I'm running this setup on a HYTiny:

```
Mecrisp-Stellaris RA 2.3.5 for STM32F103 by Matthias Koch
128 KB <g6s> 31394C34 ram/flash: 17688 18432 free ok.
```
With no I2C devices connected i2c. reports: 

```
i2c-init  ok.
i2c.
00:                         08 09 0A 0B 0C 0D 0E 0F
10: 10 11 12 13 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F
20: 20 21 22 23 24 25 26 27 28 29 2A 2B 2C 2D 2E 2F
30: 30 31 32 33 34 35 36 37 38 39 3A 3B 3C 3D 3E 3F
40: 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F
50: 50 51 52 53 54 55 56 57 58 59 5A 5B 5C 5D 5E 5F
60: 60 61 62 63 64 65 66 67 68 69 6A 6B 6C 6D 6E 6F
70: 70 71 72 73 74 75 76 77                         ok.
```

while I would expect:

```
I2c-init ok.
i2c.
00:                         -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --                         ok.
```

jcw, 2017-03-02:
Do you have a 1..10 kΩ pull-up connected to SDA? It's set to open-drain.
SCL is ok, it's push-pull, master-only, and won't support clock stretching.

jcw, 2017-03-02:
I've also inserted a delay in the default `board.fs` setup, which is probably needed now that the `io` words have been optimised.