jeelabs/embello #52: board.fs - init not found

igor-m, 2017-03-12:
While uploading the board.fs into BluePill (128k flash, latest mecrisp flashed in) I get
` ok.
!s always.fs
1> always.fs 3: Finished. Reset ☺Mecrisp-Stellaris RA 2.3.5 for STM32F103 by Mat
thias Koch
1> always.fs 13: ( always end: ) 00005060  ok.
1> always.fs 14: Redefine eraseflash.  ok.
!s board.fs
1> board.fs 3: eraseflash

Finished. Reset ☺Mecrisp-Stellaris RA 2.3.5 for STM32F103 by Matthias Koch
1> board.fs 6: ( board start: ) 00005400  ok.
7> hal.fs 14: Redefine bit.  ok.
1> board.fs 27: init not found.
Send failed.`

igor-m, 2017-03-12:
This may work?

: init ( -- )  \ board initialisation
 \ init  \ uses new uart init convention
  ['] ct-irq irq-fault !  \ show call trace in unhandled exceptions
  jtag-deinit  \ disable JTAG, we only need SWD
  72MHz
  1000 systick-hz
  hello ." ok." cr
;

jcw, 2017-03-12:
Yes, you could remove `init`, but the reason is really that Embello assumes you're using [these cores](https://github.com/jeelabs/embello/tree/master/explore/1608-forth/cores). The source for these is in the [spezial branch](https://github.com/jeelabs/mecrisp-stellaris/tree/spezial) of Mecrisp-Stelaris on GitHub. There are a few small differences (the init change is needed to support USB console I/O on F103).

igor-m, 2017-03-12:
ok
1> board.fs 26: Redefine init.  ok.
1> board.fs 36: ( board end, size: ) 00007BB4 10164  ok.
Mecrisp-Stellaris RA 2.3.5 for STM32F103 by Matthias Koch
128 KB <g6s> 3707460D ram/flash: 19356 99328 free ok.
CPU 72000000 Hz, systick 1000 Hz ok.
compiletoflash  ok.
!s 8queens.txt
compiletoram  ok.
10 queens 724 solutions, 34815 nodes ok.
14 queens 365596 solutions, 26992957 nodes ok.
10 testQ 724 solutions, 34815 nodes elapsed 68 ms ok.
14 testQ 365596 solutions, 26992957 nodes elapsed 50892 ms ok.
120MHz  ok.
1000 systick-hz  ok.
10 testQ 724 solutions, 34815 nodes elapsed 42 ms ok.
14 testQ 365596 solutions, 26992957 nodes elapsed 30530 ms ok.
128MHz  ok.
1000 systick-hz  ok.
10 testQ 724 solutions, 34815 nodes elapsed 39 ms ok.
14 testQ 365596 solutions, 26992957 nodes elapsed 28621 ms ok.

closed 2017-03-12 by igor-m