jeelabs/embello #53: BluePills with 128kB flash

igor-m, 2017-03-12:
When playing with BluePills chip C8 (128kB flash) I have to 

1. force the flash-kb to return 128 

: flash-kb ( -- u )  \ return size of flash memory in KB
\  $1FFFF7E0 h@ ;
  128 ;

2. change $10000 in the hello to $20000

: hello ( -- ) flash-kb . ." KB <g6s> " hwid hex.
  $20000 compiletoflash here -  flashvar-here compiletoram here -
  ." ram/flash: " . . ." free " ;

thus I get

 ok.
hello 128 KB <g6s> 3707460D ram/flash: 19356 99328 free  ok.

jcw, 2017-03-12:
What do you get without this change?

I wonder why this happens. Maybe there are more differences between different Blue Pills?

JohnOH, 2017-03-12:
A photograph of the Blue Pill you are using will be good to have for the records.

igor-m, 2017-03-12:
The BluePill's (and friends - black, red..) F103C8 chip returns normally 64. But most (maybe all) got 128kB in silicon.
There is only one BluePill :)
http://wiki.stm32duino.com/index.php?title=Blue_Pill

JohnOH, 2017-03-12:
We have found at least one different flavour of Blue Pill. The unusual version does not have the slot for the USB connector.