jeelabs/embello #54: Switching the mcu's clock frequency issue/improvement

igor-m, 2017-03-12:
While switching the mcu freq with words like 72MHz, 120MHz, 128MHz it always must be done via 8MHz, otherwise the PLL crashes. Here are the words for 8, 72, 120 and 128MHz (also added the systick adjustment):

@@@@@@@@@@@@@@@@@@@@
\ adjusted for STM32F103 @ 72 MHz (original STM32F100 by Igor de om1zz, 2015)
\ improvement by Igor 2017

8000000 variable clock-hz  \ the system clock is 8 MHz after reset

0 variable ticks

: ++ticks ( -- ) 1 ticks +! ;  \ for use as systick irq handler

: systick-hz ( u -- )  \ enable systick interrupt at given frequency
  ['] ++ticks irq-systick !
  clock-hz @ swap /  1- $E000E014 !  7 $E000E010 ! ;
: systick-hz? ( -- u ) \ derive current systick frequency from clock
  clock-hz @  $E000E014 @ 1+  / ;

: baud ( u -- u )  \ calculate baud rate divider, based on current clock rate
  clock-hz @ swap / ;
  
: 8MHz ( -- )  \ set the main clock back to 8 MHz, keep baud rate at 115200
  0 RCC-CFGR !                    \ revert to HSI @ 8 MHz, no PLL
  $81 RCC-CR !                    \ turn off HSE and PLL, power-up value
  $18 FLASH-ACR !                 \ zero flash wait, enable half-cycle access
  8000000 clock-hz !  115200 baud USART1-BRR !  \ fix console baud rate
  1000 systick-hz
;

: 72MHz ( -- )  \ set the main clock to 72 MHz, keep baud rate at 115200
  8MHz
  $12 FLASH-ACR !                 \ two flash mem wait states
  16 bit RCC-CR bis!              \ set HSEON
  begin 17 bit RCC-CR bit@ until  \ wait for HSERDY
  1 16 lshift                     \ HSE clock is 8 MHz Xtal source for PLL
  7 18 lshift or                  \ PLL factor: 8 MHz * 9 = 72 MHz = HCLK
  4  8 lshift or                  \ PCLK1 = HCLK/2
  2 14 lshift or                  \ ADCPRE = PCLK2/6
            2 or  RCC-CFGR !      \ PLL is the system clock
  24 bit RCC-CR bis!              \ set PLLON
  begin 25 bit RCC-CR bit@ until  \ wait for PLLRDY
  72000000 clock-hz !  115200 baud USART1-BRR !  \ fix console baud rate
  1000 systick-hz
;

\ adjusted for STM32F103 @ 120 MHz (original STM32F100 by Igor de om1zz, 2015)
: 120MHz ( -- )  \ set the main clock to 120 MHz, keep baud rate at 115200
  8MHz
  $12 FLASH-ACR !                 \ two flash mem wait states
  16 bit RCC-CR bis!              \ set HSEON
  begin 17 bit RCC-CR bit@ until  \ wait for HSERDY
  1 16 lshift                     \ HSE clock is 8 MHz Xtal source for PLL
  13 18 lshift or                  \ PLL factor: 8 MHz * 15 = 120 MHz = HCLK
  4  8 lshift or                  \ PCLK1 = HCLK/2
  2 14 lshift or                  \ ADCPRE = PCLK2/6
            2 or  RCC-CFGR !      \ PLL is the system clock
  24 bit RCC-CR bis!              \ set PLLON
  begin 25 bit RCC-CR bit@ until  \ wait for PLLRDY
  120000000 clock-hz !  115200 baud USART1-BRR !  \ fix console baud rate
  1000 systick-hz
;

\ adjusted for STM32F103 @ 128 MHz (original STM32F100 by Igor de om1zz, 2015)
: 128MHz ( -- )  \ set the main clock to 128 MHz, keep baud rate at 115200
  8MHz
  $12 FLASH-ACR !                 \ two flash mem wait states
  16 bit RCC-CR bis!              \ set HSEON
  begin 17 bit RCC-CR bit@ until  \ wait for HSERDY
  1 16 lshift                     \ HSE clock is 8 MHz Xtal source for PLL
  14 18 lshift or                  \ PLL factor: 8 MHz * 16 = 128 MHz = HCLK
  4  8 lshift or                  \ PCLK1 = HCLK/2
  2 14 lshift or                  \ ADCPRE = PCLK2/6
            2 or  RCC-CFGR !      \ PLL is the system clock
  24 bit RCC-CR bis!              \ set PLLON
  begin 25 bit RCC-CR bit@ until  \ wait for PLLRDY
  128000000 clock-hz !  115200 baud USART1-BRR !  \ fix console baud rate
  1000 systick-hz
;

@@@@@@@@@@@@@
Mecrisp-Stellaris RA 2.3.5 for STM32F103 by Matthias Koch
128 KB <g6s> 3707460D ram/flash: 19356 98304 free ok.
CPU 72000000 Hz, systick 1000 Hz ok.
compiletoflash  ok.
!s 8queens.txt
12 queens 14200 solutions, 841989 nodes ok.
13 queens 73712 solutions, 4601178 nodes ok.
12 testQ 14200 solutions, 841989 nodes elapsed 1594 ms ok.
13 testQ 73712 solutions, 4601178 nodes elapsed 8694 ms ok.
120MHz  ok.
12 testQ 14200 solutions, 841989 nodes elapsed 957 ms ok.
13 testQ 73712 solutions, 4601178 nodes elapsed 5216 ms ok.
128MHz  ok.
12 testQ 14200 solutions, 841989 nodes elapsed 897 ms ok.
13 testQ 73712 solutions, 4601178 nodes elapsed 4889 ms ok.

jcw, 2017-03-12:
Ooh - nice! Are you over-clocking? That's a quite a speedup...

igor-m, 2017-03-12:
Yes, that is Overclocked, the performance in 8Queens at 128MHz is higher than with the F407 @168MHz (due to less flash wait states). It works 128M fine here, on BluePill as well as on 103ZET6.

igor-m, 2017-03-12:
I've sent you the new hal with the improvements for 103 and friends, you may try..

JohnOH, 2017-03-12:
Nice, I wonder what it looks like in C.

jcw, 2017-05-01:
I like the speed bump we can get for free, but it's not something to include by default in all builds, IMO - my suggestion would be to add the overclocking variants to a new `hal-oc.fs` file. It also looks like quite a bit of code in all these variants could be refactored and combined.

FWIW, I've adjusted the std 72 MHz word to switch back to 8 MHz first, as you suggest.

commit 6ba8e2e1 (2017-05-01): always switch off PLL before goint to 72 MHz - closes #54
closed 2017-05-01 by jcw