jeelabs/embello #55: 8 Queens benchmark

igor-m, 2017-03-12:
0 variable solutions
0 variable nodes
 
: bits ( n -- mask ) 1 swap lshift 1- ;
: lowBit  ( mask -- bit ) dup negate and ;
: lowBit- ( mask -- bits ) dup 1- and ;
 
: next3 ( dl dr f files -- dl dr f dl' dr' f' )
  not >r
  2 pick r@ and 2* 1+
  2 pick r@ and 2/
  2 pick r> and ;
 
: try ( dl dr f -- )
  dup if
    1 nodes +!
    dup 2over and and
    begin ?dup while
      dup >r lowBit next3 recurse r> lowBit-
    repeat
  else 1 solutions +! then
  drop 2drop ;
 
: queens ( n -- )
  0 solutions ! 0 nodes !
  -1 -1 rot bits try
  solutions @ . ." solutions, " nodes @ . ." nodes" ;
  
: testQ ( N -- ) millis @ swap queens millis @ swap - ."   t= " . ." ms" ;
  
 \ 8 queens 
 \ 8 queens 92 solutions, 1965 nodes ok.
 \ 12 queens 14200 solutions, 841989 nodes ok.
 \ 14 queens 365596 solutions, 26992957 nodes ok.
 \
 \ STM32F407, 168MHz clock, serial 115k2:
 \ ==============================================
 \ 8 testQ 92 solutions, 1965 nodes  t= 4 ms ok.
 \ 10 testQ 724 solutions, 34815 nodes  t= 43 ms ok.
 \ 12 testQ 14200 solutions, 841989 nodes  t= 1002 ms ok.
 \ 13 testQ 73712 solutions, 4601178 nodes  t= 5457 ms ok.
 \ 14 testQ 365596 solutions, 26992957 nodes  t= 31936 ms ok.
 \ 16 testQ 14772512 solutions, 1126417791 nodes  t= 1331997 ms ok.

igor-m, 2017-03-12:
: testQ ( N -- ) millis swap queens millis swap - ." t= " . ." ms" ;

jcw, 2017-03-15:
Neat! And rather surprising that an M3 can outperform an M4 (both tests RAM-based, I assume).
It might be useful to find out the F407 Mecrisp version, maybe there have been tweaks since then.

igor-m, 2017-03-16:
Most probably 2.1.3, edited above source.

jcw, 2017-03-16:
Ah, and the other thing to note is non-RA vs RA - that could also explain the difference.