jeelabs/embello #56: low-power sleep example hangs after first loop (JNZ-rev3)

doewie, 2017-03-15:
I'm running this config on JNZ-rev.3
```
Mecrisp-Stellaris RA 2.3.5 with M0 core for STM32L053C8 by Matthias Koch
64 KB <jz3> 3239023E ram/flash: 6012 22272 free ok.
```
Running the original blink example from the commandline is OK
```
: lp-blink ( -- )  only-msi  begin  stop1s led iox!  again ;
lptim-init lp-blink
```
But running following testfile and run with` lptim-init lp-blink` just toggles the LED once and then stops.
```
\ file: exp.fs

\ blink example
: lp-blink ( -- ) 
	only-msi
	begin
		stop1s 
		led iox!
	until
	;
```

jcw, 2017-03-15:
I think you need to change `until` (which wants a flag on the stack) to `again`.

doewie, 2017-03-15:
Grrr... Totally overlooked that stupid typo. 

But I'm still struggling with this low-power. If I run the code below the LED indeed toggles 5 times, but then connection with the JNZ is lost and can only be restored with !reset. Or is this a folie issue?
```
: lp-loop ( -- )
	lptim-init
	only-msi
	5 0 do
		stop1s 
		led iox!
	loop
	;
```

jcw, 2017-03-15:
That's because with only MSI running, you don't have a clock for the serial port anymore.
You can get serial back by calling `hsi-on` after your loop exit.