jeelabs/embello #60: stm32f4/io.fs incorrectly using stm32f1 register layout

pelrun, 2017-04-25:
The F4 has the io mode and PP/OD state in separate registers, unlike the F1 where they're interleaved in CRL/CRH. The constants and the io-mode! word are copied almost verbatim from the F1 code and are therefore not working.

jcw, 2017-05-01:
Maybe the [stm32l0/io.fs](https://github.com/jeelabs/embello/blob/master/explore/1608-forth/flib/stm32l0/io.fs) code would be a better starting point? I haven't used the F4 much, I admit.

pelrun, 2017-05-01:
I managed to put together a working replacement myself, but since it's pretty much the first real Forth I've written it's pretty suboptimal :) - I'll test the L0 code and if it's good (or only needs minor tweaks) I'll report back.

PR #61 (2017-05-08): Replace broken stm32f4/io.fs with working one based off stm32l0 version.
Fixes #60, at least in initial testing.
closed 2017-05-05 by jcw