jeelabs/embello #62: i2c on stm32f1 reading 1-byte replies

smurfix, 2017-05-05:
Hi,
I seem to be unable to read a 1-byte reply on i2c.
Looking at i2c-xfer, that case seems to be a no-op
```      1 of endof                ( cnt = 1 )
```
which can't possibly be correct.

jcw, 2017-05-05:
Hm, odd indeed - that's a new driver, we'll have to ask @pragtich for help on this one.

You can switch to the bit-banged version, i.e. use the `flib/any/i2c-bb.fs` driver instead until this issue gets resolved.

pragtich, 2017-05-05:
Oops,
You're right, of course. I had postponed implementing that option till I understood the rest. 
I might have some time to fix it over the weekend, shouldn't be too complicated.
I clearly need a test suite :-)

PR #63 (2017-05-06): Add i2c-xfer for N=1
This should fix #62 by adding i2c-xfer behaviour for N=1.
This was an oversight on my part: I had waited implementing this one because it needed some register twiddling inside EV6. Eventually, I think this is a simple way to solve it.

Would really appreciate some feedback if this works as expected.

Next time, I'll leave whitespace cleanup out of it :-)
closed 2017-05-06 by jcw