jeelabs/embello #69: I2C oled displays working?

pragtich, 2017-05-09:
I was trying to do some further testing on the hardware I2C interface for the STM32f103 yesterday evening. I am on commit ca06024.
 To my consternation, I cannot for the life of me seem to get either of my 2 cheap Chinesium OLED displays to work anymore.
They do report their presence at address $3C using the probe `i2c.`, but they never turn on. 
Of course, these are quite low quality units, and they have seen some mechanical stresses recently so I might just have damaged them.
Are others seeing the same, by any chance? Or are your oled displays working?

jcw, 2017-05-09:
Funny you mention that - same here. I've not yet checked with the bit-banged version, but there seems to have been some regression. I too can detect the device on the bus. Could it be that we need external pull-ups when using I2C in hardware mode?

JohnOH, 2017-05-09:
I didn't think that external pull-ups were optional.

jcw, 2017-05-09:
The bit-banged version enables (weak) internal pull-ups.

pragtich, 2017-05-09:
I have the pull ups in place.

JohnOH, 2017-05-09:
What resistor values did you use?

pragtich, 2017-05-09:
I'll check tonight.
Fyi, the hw i2c driver configures the pins as open drain, no pulling up from the chip.

tve, 2017-05-09:
Mhh, I just reloaded a JNZ-v1 with some of my code that uses the ssd1306 OLED and it initializes and display the JL logo just fine. 
```
16MHz 1000 systick-hz  ok.
lcd-init  ok.
clear  ok.
display  ok.
show-logo  ok.
```
This works using both the bb i2c driver and the HW version. (L052 processor, not F103)

jcw, 2017-05-09:
To clarify: in my case, I am referring to the F103.

pragtich, 2017-05-10:
@tve  , thanks for the feedback. Since the L052 has a different I2C implementation, I think there's no relation.

I was doing some probing with the cheap logic analyzer last night. And reading the SSD1306 datasheet. Here's a partial update.

The key is in the datasheet:

> It should be noticed that the ITO track resistance and the pulled-up resistance at “SDA” pin becomes
> a voltage potential divider. As a result, the acknowledgement would not be possible to attain a valid
> logic 0 level in “SDA”.
> “SDAIN” and “SDAOUT” are tied together and serve as SDA. The “SDAIN” pin must be connected to
> act as SDA. The “SDAOUT” pin may be disconnected. When “SDAOUT” pin is disconnected, the
> acknowledgement signal will be ignored in the I2C-bus

So the OLED does not ACK the transmissions for either one of 2 possible reasons.

That's also what I see in the logic analyzer: always NAK. I recorded the sequence of communication during `lcd-init`, so sending a lot of command byte pairs `$80  command`. I see the address byte and first command byte sent without problem ($c3 $80). But both without receiving ACK from the display. And then a restart gets generated for the next transmission, presumably because of the NAK. 

Next step is to confirm that the NAK is causing the failure of the second >i2c. 
And then to find out
how to get the driver to continue without ack, when we want that to happen (and not when we don't).
I'm also curious how come the `i2c.` detects the oled without the ACK being given...

jcw, 2017-05-10:
But then... why did the OLED work fine before? With the bit-banged implementation, that is.

pragtich, 2017-05-10:
@jcw, can you confirm that the bit-banged implementation does not work for the oled displays at the moment?
When I was testing the hardware driver, I was travelling and I did not test on the SSD1306, just on a few sensors that I had with me. I cannot say that I've ever seen the hardware driver work for the oleds. Except responding positively to the probe.

Perhaps the differing pull-up situation allows the ACK to get through in the bit banged version? (Just a speculation). It's tricky with these cheap logic analyzers too, because they also load the bus (mine seems to be pulling up).

jcw, 2017-05-10:
I'll investigate later today.

jcw, 2017-05-10:
Ok, with `10 constant i2c.delay` defined before `include ../flib/any/i2c-bb.fs`, I can confirm that I2C is working here as before, i.e. using the original bit-banged code. I don't think I have external pull-ups (they might be on the OLED board, dunno).

To test the OLED, I used `lcd-init show-logo`.

pragtich, 2017-05-11:
@jcw, that's good to know. Makes it more likely that the problem is in the f103 driver, and not in the changes to the ssd1306 library.

My assumption in programming the f103 hardware driver, had always been to follow the I2C standard, so it selects open drain outputs and assumes external pull-ups. This is not hard and fast: for small systems it is probably valid to use the internal pull-ups instead. Once the bug is fixed, we should come up with a plan (make it configurable, maybe?)

 Because of the chip-on-glass technology, I have learned that there is a certain (unspecified??) resistance in the lines to the SSD1306, which makes the reception of the ACK unreliable. That explains why the problem is specific to the display, and not the other I2C chips.

The strangest thing, is that we tend to get an ACK for the probe, but usually not the communication, not all the time at least. Probably a timing issue? I don't have a 'scope, so hard to debug. I think I'll try some larger values for the pull ups (am using 1.2kOhm now), to test the theory. 

The STM32f103 reference manual does not really specify how the device responds to NAK on transmit, except mentioning that the communication should be reset. My hypothesis is, that once the AF bit is set, transmission stops. I would test this by temporarily putting a `i2c-AF-0` at the beginning of `>i2c`, or before both the `>i2c` commands in `lcd!c`. Perhaps that already allows the transmission to continue.

I don't expect to have time to test this until tomorrow evening or Saturday, feel free to give it a go if you have the chance. Otherwise, I'll keep you posted.

jcw, 2017-05-11:
From what I can tell, internal pull-ups can be enabled even when a h/w function like I2C is used. I'll try later today if that is enough to get the h/w driver going w/ my OLED.

[image]

pragtich, 2017-05-11:
That might well work: the pull ups on the chip are quite weak:
[image]

jcw, 2017-05-11:
Ok, enabling the pull-ups appears to make a difference:

```
imode-pull pb6 io-mode!  ok.
imode-pull pb7 io-mode!  ok.
pb6 ios!  ok.
pb7 ios!  ok.
lcd-init  ok.
show-logo  ok.
demo  ok.
```

I got a logo to appear, although there were some issues with pixel offsets. It isn't consistent though, on retry nothing happened. Those pull-ups are definitely very weak - 3kΩ would be better (≈1 mA @ 3.3V).

pragtich, 2017-05-13:
Yes, well.. At the moment the driver is trying to go at 400 kHz, so the weak pullups are going to be way too slow, probably. That would explain the intermittent behaviour. I should choose a slower default speed, that allows lower power loss through the pullups and should be sufficiently fast for most use cases. And will allow much larger tolerance on the pull up resistance, which is also a good thing. This will also reduce the risk of NAK with the SSD1306.

Some intermediate results from yesterday:
- Adding `i2c-AF-0` to `>i2c`: no display
- Adding a delay to the start of `>i2c`: perfect display, down to 20 us even
- Slowing bus to 100 kHz and 20us delay: no display 

The last one needs some more work: probably a larger pull up would help?

More to explore... :-)

jcw, 2017-05-13:
Good progress - from other tests a while back (completely diff s/w), my impression was that these OLEDs can go far above 400 KHz (not surprising perhaps - since it can also handle much faster SPI).

pragtich, 2017-05-14:
@jcw, do you have any explanation for this phenomenon?
In all cases I am loading a version of i2c.fs and ssd1306.fs into ram, and call `i2c-init i2c.`, which invariably shows the display. Then I test by calling `lcd-init show-logo` and sometimes it will display nothing, sometimes it will work fine and sometimes it will give an image, but with some rows shifted.

- `20 us` added to the beginning of `>i2c`: everything good
- `20 us` or even `50 us` added before both calls to `>i2c` in `lcd!c`: no display

How can that be?!

jcw, 2017-05-14:
The rows-shifted is what I've been seeing as well. Not sure, it will probably require debugging with a scope or logic analyser - there are $15'ish LAs on eBay, if you're interested in that. I can fire up mine here, but it may take me a while to get around to it. Will report back here if/when I do.

pragtich, 2017-05-14:
Thanks, I have one of those logic analysers, will try with that.

pragtich, 2017-05-15:
Think I found it: i2c-xfer with argument 0 calls i2c-stop too early. That stifles the last byte in case of multiple bytes sent in succession. The solution is simple, adding the correct EV8_2. Will make a pull request tomorrow.

jcw, 2017-05-15:
Aha, good catch!

pragtich, 2017-05-16:
I won't take the compliment: it's my own errors that caused this mess.

In final testing of my patch, I've found several more issues in the behaviour of i2c-xfer, >i2c and i2c>... Will try and fix them...

jcw, 2017-05-16:
It's still a good job in my book - I had a heck of a tough time getting h/w I2C going on L052, and skimmed over many details.

PR #75 (2017-05-16): Add i2c-EV8_2, fixes #69
I'll submit this PR, which fixes #69. The last byte of multibyte transfers was being clobbered by `0 i2c-xfer`. That is working correctly now, and gets my Oled to light up and respond to commands like `show-logo` and `demo`.

- I still have issues with the oled: only the odd rows are lighting up and everything is stretched out in the vertical direction. This is apparently a common issue, and should be easily fixed?
- In testing this patch, I uncovered several issues in multibyte reads using `i2c>`. I'll start a new issue for that.
commit 528d42ca (2017-05-16): Add i2c-EV8_2, fixes #69
commit 6aba48fa (2017-05-16): Merge pull request #75 from pragtich/f103-i2c-multibyte-fix

Add i2c-EV8_2, fixes #69
closed 2017-05-16 by jcw