jeelabs/embello #73: usbserup make issue

tj57, 2017-05-10:
I can't compile usbserup, I got message:
> t@ubuntu ~/embello/tools/usbserup $ make
make: *** No rule to make target 'usbserup.elf', needed by 'elf'. Stop.

How to compile it ?

jcw, 2017-05-10:
This is what I get with `make -n`:

```
#printf "  CC      usbserup.c\n"
arm-none-eabi-gcc -std=gnu99 -Os -g -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Wmissing-prototypes -Wstrict-prototypes -fno-common -ffunction-sections -fdata-sections -MD -Wall -Wundef -I./libopencm3/include -DSTM32F1 -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd -o usbserup.o -c usbserup.c
#printf "  LD      usbserup.elf\n"
arm-none-eabi-gcc --static -nostartfiles -L./libopencm3/lib -T./stm32f100xb-high.ld -Wl,-Map=usbserup.map -Wl,--gc-sections -mthumb -mcpu=cortex-m3 -msoft-float -mfix-cortex-m3-ldrd usbserup.o -lopencm3_stm32f1 -Wl,--start-group -lc -lgcc -lnosys -Wl,--end-group -o usbserup.elf
gcc -o bin2h bin2h.c
#printf "  OBJCOPY usbserup.bin\n"
arm-none-eabi-objcopy -Obinary usbserup.elf usbserup.bin
./bin2h usbserup.bin >data.h
arm-none-eabi-size usbserup.elf
```

Does your output look the same?

tj57, 2017-05-11:
Not

> t@ubuntu ~/embello/tools/usbserup $ cat Makefile 
> BINARY = usbserup
> 
> OPENCM3_DIR=./libopencm3
> LDSCRIPT = ./stm32f100xb-high.ld
> CFLAGS = -std=gnu99
> 
> default: all data.h
> 	arm-none-eabi-size usbserup.elf
> 
> data.h: bin2h $(BINARY).bin
> 	./bin2h $(BINARY).bin >$@
> 
> bin2h: bin2h.c
> 	gcc -o $@ $<
> 
> include ./libopencm3.target.mk
> t@ubuntu ~/embello/tools/usbserup $ make -n
> make: *** No rule to make target 'usbserup.elf', needed by 'elf'. Stop.
>

tve, 2017-05-11:
Same here:
```
> make
make: *** No rule to make target 'usbserup.elf', needed by 'elf'.  Stop.
```
This is on an x64 system running ubuntu. Is there something I need to install?

tve, 2017-05-11:
THere's a `libopencm3` subdir, which is empty. Did you perhaps forget to check-in the files that go there? Or where do I get libopencm3 from and where do I put it?

tj57, 2017-05-11:
Libopencm3 is notempty, I used.
> git submodule update --init

Ok, I have found issue, missing make in libopencm3 subdirectory.
In tutorial should be:
> git clone
> git submodule update --init
> cd  libopencm3 
> make
> and finally make in usbserup directory

jcw, 2017-05-11:
Aha, that explains it - many thanks, I'll update the README!

commit 12cbb37e (2017-05-11): better build instructions, fixes #73
closed 2017-05-11 by jcw