jeelabs/embello #76: STM32f1 hardware I2C data reads data incorrectly

pragtich, 2017-05-16:
In testing for #69, I uncovered at least one issue with the hardware I2C driver for STM32F103.

When reading multiple bytes, some data gets corrupted.

A simple demonstration, using a register of the ADS1113 A/D converter as a scratch pad (and with `i2c>` temporarily modified to read two bytes):

`
 $48 i2c-addr 3 >i2c $34 >i2c  $12 >i2c 0 i2c-xfer  
$48 i2c-addr 3 >i2c $34 >i2c  $12 >i2c 0 i2c-xfer    ok.
$48 i2c-addr 3 >i2c 2 i2c-xfer i2c> swap h.2 h.2
$48 i2c-addr 3 >i2c 2 i2c-xfer i2c> swap h.2 h.2 3424 ok.
`
So: you write $3412 and read back $3424. This second byte is consistently shifted left by 1 bit. Other weird stuff is also happening, but let's focus on this first.

My logic analyzer agrees with me:
[image]

pragtich, 2017-05-16:
Thank you, STM, for making errata but not correcting the manual. 

[Time for some study](http://www.st.com/content/ccc/resource/technical/document/errata_sheet/7d/02/75/64/17/fc/4d/fd/CD00190234.pdf/files/CD00190234.pdf/jcr:content/translations/en.CD00190234.pdf)

pragtich, 2017-05-17:
So the workaround for the shifted-last-byte issue is to pull down SCL manually, in order to delay the Stop event. [I have it working for N=2, N>2 still needs some work](https://github.com/pragtich/embello/blob/hwi2c-corrupt-recv/explore/1608-forth/g6u/i2c.fs#L198).

However, there are so many race conditions mentioned in the Errata, where "_**problems may be encountered such as receiving an extra byte, reading the same data twice or missing data**_.", that I am wondering weither I should try implementing the DMA solution. Sounds like it could be fun...

jcw, 2017-05-17:
Which begs the question: if h/w I2C is so hard on F103, should we use the bit-banged driver by default instead? Especially now that it supports clock-stretching.

While DMA sounds like an interesting challenge, I'm not sure it buys us that much - even at 400 KHz, I2C data transfers come at an "almost leisurely" pace of max 1 byte per 22.5 µs. Also: are you sure the DMA approach would be less error-prone, if the I2C h/w itself is weak in the first place?

pragtich, 2017-05-17:
If I can trust the Errata, the DMA code would be more reliable. Because it guarantees timings a bit closer (transfer _speed_ is not the reason for using DMA, rather it is transfer _timing_). And it would definitely cut back on the code quite dramatically. But the previous experience has taught me to be cautious: the I2C implementation in the F103 is just not really nice. How can it be that a bit banged interface is more concise and more elegant, and performs as well for all intent and purpose? I only see benefit of the hw interface if you use interrupts and multitasking intensively.

So, I would not be sad if you chose to use the bit banged driver as a default. It works fine, especially now that it supports clock stretching. 

I have now been triggered enough to want to try to get the DMA implemention working, just for the heck of it. But don't wait for it :-)
It could allow a different I2C API, relying on callbacks. But I get the feeling that simplicity is better, for wireless nodes.

[image]

jcw, 2017-06-05:
Haha - lovely sequence of how IT usually works :)

pragtich, 2017-06-28:
So I have a working prototype of a DMA driven solution for STM32F103 hardware I2C. It clearly still suffers from all the problems of not being at all easy to understand, and it needs a lot of cleaning up. But it seems to be doing the job. Data is not arriving corrupted anymore, at least. The DMA engine is actually really nice, fun to play with and really fast. And it does a lot of the work automatically.

Too bad that the I2C engine in these chips is not very nice to work with. It amazes me how much boilerplate is needed to get things to run.

I'll put this here for your information and try and clean it up with low priority in the coming weeks. 

For most applications, I guess that the bit banged interface is probably the better choice, as we concluded before.

[Here's the prototype](https://github.com/pragtich/embello/blob/dma-hw-i2c/explore/1608-forth/dmai2c/i2c.fs)

jcw, 2017-06-28:
Congrats on getting it working!

Yeah, h/w I2C is a messy proposition, but it's still very nice to see what it takes to jump through all the hoops, if only to demonstrate the complexity of the I2C and the DMA hardware. I hope you'll submit it as pull request at some point, so that we can add it to the jeelabs/embello repo.

I agree, DMA is fun - lots of things start to happen magically once you get it going.

pragtich, 2017-08-01:
@jcw, I am working on getting the I2C driver with DMA ready for a pull request. I am happy with how it is working, and it actually rebased well, so that made my day.

One thing that I needed to do, is recompile Mecrisp in order to get the dma interrupt vectors available. For my own testing, I made a `g6u` with my own mecrisp binary, and I will probably end up making one for `g6s` too, in order to test that case (especially changing clock frequencies).

Question: how would you like me to submit the Mecrisp portion of the changes? Should I just submit the changes to the source, or do you want one or several binaries included? Personally, I would prefer that you did the compiling, because I'm not sure that I have a complete understanding of all the things that might break.

jcw, 2017-08-01:
Great progress! - Re interrupt vectors: I've had some discussion with Matthias Koch about this, and realised that not each vector needs to be split off separately to be usable. The catch-all interrupt can also be used, by including a dispatch on the current interrupt number. So technically, it should be possible to run I2C w/ DMA without altering the Mecrisp build.

Having said that, the Mecrisp cores being used in embello (both L052 and F103) are all built from the "spezial" branch - see https://github.com/jeelabs/mecrisp-stellaris/compare/master...spezial - if the change does not increase the code size too much, we could consider adding it to the spezial branch. There's a limit to this, because we may need to make other changes / bug fixes later, and need to remain under 20 KB for the RA core. Plus, there's no easy way back: once added, it's hard to revert such an extension once it finds its way into many installations.

From your remarks, I gather that this driver is for the F103?

pragtich, 2017-08-01:
Thanks for the compliment, I'm actually really feeling like a chemist now: this project has escalated to quite low level stuff, lots of learning required. Once done, I'll try and actually get a node up :-)

You are right, this driver is specific to the STM32F103 (I have no other boards and am not aware of other compatible I2C peripherals), and pretty much provides a workaround to the data integrity issues that the current hardware I2C driver has due to defects in the chip. It uses the DMA peripheral, because that allows interrupts to perform the required actions at stricter timing than polling would. At least, that's what the errata sheet promises, and up to now my testing has been successful. This makes me somewhat leery to think about using a dispatch scheme: do we not run the risk of unpredictable timing? (Perhaps not, if we disable interrupts in the dispatcher?)

I can understand the issue with adding too many irq vectors, which may only be useful to a small subset of users. Perhaps we may only add the dma1 channel 6 and 7 vectors, these are the only 2 that I need? Would that be acceptable?

jcw, 2017-08-01:
Dispatch: it's simply a test inside the catch-all interrupt, which runs with interrupts off. BTW, I've suggested a change for Mecrisp to expose all irq vecs as a single word, iso the current set (although chances for such an invasive change to be accepted by @Mecrisp may be slim).

Re the catch-all interrupt: there's a hook variable called `irq-collection` for it - you can set it to point to your own handler and then use `ipsr` to get the exception number and switch on it. See <https://github.com/jeelabs/mecrisp-stellaris/blob/master/mecrisp-stellaris-source/common/interrupts-common.s#L145>. I don't think you'll notice any negative effects from the minor slowdown, this is just for interrupts, not for each DMA byte transfer, and the `case` (or `if`) takes little time.

pragtich, 2017-08-02:
Thanks for the suggestion, did not know about this. I'll give it a go!

pragtich, 2017-08-03:
Okay, got that up and running in no time. So the driver now works from a stock Mecrisp image.
I guess I should try and call a pre-existing irq-collection function, if it is <> `unhandled`, just in case someone else also needs an interrupt caught. I am thinking to save the existing value of `irq-collection` into a variable, and calling it at the end of my own dispatch function. What do you think?

jcw, 2017-08-03:
Neat! Yes, for library code that's probably wise - I don't think there's anything in Mecrisp to help with it, so you'll have to bring your own var :)

There _is_ one feature which might help de-clutter slightly - check out [quotations](https://github.com/jeelabs/mecrisp-stellaris/blob/master/common/quotations.txt), which are usually included in the embello setup. It's a way to create a function on the fly, without giving it a name.

PR #87 (2017-08-26): Stm32f1 hwi2c
Hi all,

I started this as a challenge to learn programming the STM32F103 (blue pill) and Forth. Since there are issues with the hardware of the chip, [DMA transfer is recommended](http://www.st.com/content/ccc/resource/technical/document/errata_sheet/7d/02/75/64/17/fc/4d/fd/CD00190234.pdf/files/CD00190234.pdf/jcr:content/translations/en.CD00190234.pdf). This hardware driver uses the DMA engine in order to circumvent the problems with the chip.

I hope it will work for you as it works for me, and I'll move attention to actually getting a node up and running :-)

Please review & l [... cut]
closed 2017-08-26 by jcw