jeelabs/embello #78: STM32F103 and USB

agsb, 2017-05-19:
Hi,
Great Job with the Blue Pills! 
But I'm not getting a stm32F1 (0410) work  with usb connection. 
I done the upload !u 3 with folie 2.12 and a FTDI and boot0 at 1, ok all good, 
change boot0 to 0, reset, and board talks forth with folie 2.12, ok al good. 
When I unplug FTDI and plug a USB cable, reset and then, no talk! 
Any tip to check if USB terminal forth is on ? That PA12 is going up by software ?
what i forgot ?
best regards
Alvaro

jcw, 2017-05-19:
Are you passing the "-r" (raw) flag to folie?

jcw, 2017-05-19:
Did you upload the the proper flash image, which includes the USB driver?
See https://github.com/jeelabs/embello/tree/master/explore/1608-forth/suf

agsb, 2017-05-19:
No, I upload the !u <3> option from folie 2.12, I will try that of
suf/usb-common.hex, next week and cover the results.
Thanks,

agsb, 2017-06-05:
hello,

trying again.

using a ubuntu 17.04/64bits, blue pill with boot0 at one and boot1 at zero,
serial CP2102, tx-rx as in http://jeelabs.org/article/1613a/, last v2
folie-linux with -r and !u
https://raw.githubusercontent.com/jeelabs/embello/master/explore/1608-forth/suf/usb-common.hex
got this:

fetching ok, crc 07b6
48128b .....
and a monotonic sequence of ..........

I push tha reset button at blue pill, and nothing happens

things to do ?

thanks for any advice or tips

[]s

Alvaro

agsb, 2017-06-05:
> com/jeelabs/embello/master/explore/1608-forth/suf/usb-common.hex
> forth/suf

hello,
update:

with folie-linux64 (md5sum 3c55f95c83be90a9217ead9086168824) and press
reset button at bluepill:

panic: runtime error: invalid memory address or nil pointer dereference
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x533a0b]
goroutine 19 [running]:
go.bug.st/serial%2ev1.(*unixPort).getModemBitsStatus(0x0, 0x451ef6,
0x40f92f, 0xc4200dd708)
    /Users/jcw/Go/src/go.bug.st/serial.v1/serial_unix.go:376 +0x2b
go.bug.st/serial%2ev1.(*unixPort).SetDTR(0x0, 0xc4200dd701, 0xc4203c8020,
0xc4200dd760)
    /Users/jcw/Go/src/go.bug.st/serial.v1/serial_unix.go:97 +0x2f
main.boardReset(0xc420480300)
    /Users/jcw/Go/src/github.com/jeelabs/folie/serial.go:74 +0x6c
panic(0x6692c0, 0x7fec00)
    /usr/local/Cellar/go/1.8.1/libexec/src/runtime/panic.go:489 +0x2cf
go.bug.st/serial%2ev1.(*unixPort).getModemBitsStatus(0x0, 0x4ad2d7,
0x7ca040, 0xc42000c040)
    /Users/jcw/Go/src/go.bug.st/serial.v1/serial_unix.go:376 +0x2b
go.bug.st/serial%2ev1.(*unixPort).SetDTR(0x0, 0x1, 0x0, 0x0)
    /Users/jcw/Go/src/go.bug.st/serial.v1/serial_unix.go:97 +0x2f
main.boardReset(0xc4200dd901)
    /Users/jcw/Go/src/github.com/jeelabs/folie/serial.go:74 +0x6c
main.connectToTarget()
    /Users/jcw/Go/src/github.com/jeelabs/folie/upload.go:147 +0x2a
main.Uploader(0xc4204a8000, 0xbc00, 0xe000)
    /Users/jcw/Go/src/github.com/jeelabs/folie/upload.go:41 +0x1b2
main.wrappedUpload(0xc42047d200, 0x2, 0x2)
    /Users/jcw/Go/src/github.com/jeelabs/folie/wrapped.go:132 +0x522
main.SpecialCommand(0xc4203c27e0, 0x61, 0x6b4471)
    /Users/jcw/Go/src/github.com/jeelabs/folie/serial.go:225 +0x70b
main.SerialDispatch()
    /Users/jcw/Go/src/github.com/jeelabs/folie/serial.go:180 +0x47b
created by main.blockUntilOpen
    /Users/jcw/Go/src/github.com/jeelabs/folie/serial.go:107 +0x306

jcw, 2017-06-05:
Not sure this helps, but you need to use the serial port to upload the `usb-common.hex` file, then after reset Mecrisp will switch to USB as its console.

PS. I'll be away for a while, so can't help much further a.t.m. - please feel free to add whatever info you have and I'll revisit once I get back.

agsb, 2017-06-05:
hello, seens someone reversed the rx-tx at cp2102, so I try another one and
all goes ok.

Please confirm if the code for usb-common.hex use the internal 1.5k ohm
resistor up in software?

https://www.eevblog.com/forum/projects/stm32-usb/?action=dlattach;attach=95498;image

thanks a lot!
[]s

Alvaro

agsb, 2017-06-06:
Hello,

using a external 1k5 resistor at PA12 (D+) to Vcc make the device shows

[  438.372085] usb 3-2: USB disconnect, device number 2
[  448.612035] usb 3-2: new full-speed USB device number 3 using uhci_hcd
[  448.804055] usb 3-2: New USB device found, idVendor=0483, idProduct=5740
[  448.804058] usb 3-2: New USB device strings: Mfr=1, Product=2,
SerialNumber=3
[  448.804060] usb 3-2: Product: Forth Serial Port
[  448.804061] usb 3-2: Manufacturer: Mecrisp (STM32F10x)
[  448.804063] usb 3-2: SerialNumber: 30323217
[  448.807123] cdc_acm 3-2:1.0: ttyACM0: USB ACM device

And folie-linux64 -r       seems work.

[]s

On Mon, Jun 5, 2017 at 6:54 PM, Alvaro Barcellos <barcellos.alvaro@gmail.com

> hello, seens someone reversed the rx-tx at cp2102, so I try another one
> and all goes ok.
>
> Please confirm if the code for usb-common.hex use the internal 1.5k ohm
> resistor up in software?
>
> https://www.eevblog.com/forum/projects/stm32-usb/?action=
> dlattach;attach=95498;image
>
> thanks a lot!
> []s
>
> Alvaro
>
> On Mon, Jun 5, 2017 at 1:30 PM, Jean-Claude Wippler <
>
>> Not sure this helps, but you need to use the serial port to upload the
>> usb-common.hex file, then after reset Mecrisp will switch to USB as its
>> console.
>>
>> PS. I'll be away for a while, so can't help much further a.t.m. - please
>> feel free to add whatever info you have and I'll revisit once I get back.
>>
>>
>
>
>
>
>
> *"Sling and move continents, a rock at time.*"

agsb, 2017-06-06:
Hello,

To pull-up internal 1k5 ohms at PA12, from
8.5.2 SYSCFG peripheral mode configuration register (SYSCFG_PMC), page 219,
http://www.st.com/content/ccc/resource/technical/document/reference_manual/cc/f9/93/b2/f0/82/42/57/CD00240193.pdf/files/CD00240193.pdf/jcr:content/translations/en.CD00240193.pdf

*diff f-common.fs f-common.fs.old 29d28<     1 bit $40010004 bis! \ set
PA12 internal 1k5 Ohms resistor for USB device recognition, USB_PU at
SYSCFG_PMC *
thanks
[s]

mvcorrea, 2017-06-10:
hi there,
using an st-linkv2 to program a bluepill with "embello/explore/1608-forth/suf/usb-common.bin"
on osx...

boot0 == 1 && boot1 == 0

`% st-flash --reset write usb-common.bin 0x8000000
st-flash 1.3.1
2017-06-10T12:58:45 INFO src/common.c: Loading device parameters....
2017-06-10T12:58:45 INFO src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2017-06-10T12:58:45 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2017-06-10T12:58:45 INFO src/common.c: Attempting to write 48128 (0xbc00) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0800b800 erased
2017-06-10T12:58:46 INFO src/common.c: Finished erasing 47 pages of 1024 (0x400) bytes
2017-06-10T12:58:46 INFO src/common.c: Starting Flash write for VL/F0/F3 core id
2017-06-10T12:58:46 INFO src/flash_loader.c: Successfully loaded flash loader in sram
 46/47 pages written
2017-06-10T12:58:48 INFO src/common.c: Starting verification of write complete
2017-06-10T12:58:49 INFO src/common.c: Flash written and verified! jolly good!`

then with folie..

` ./folie-macos64
Folie v2.12
Select the serial port:
  1: /dev/cu.Bluetooth-Incoming-Port
  2: /dev/cu.usbmodemD5222A31
2
Enter '!help' for additional help, or ctrl-d to quit.
[connected to /dev/cu.usbmodemD5222A31]
!u
These firmware images are built-in:
  1: F103-BMP         50096b  crc:F87A
  2: F103-Blink         724b  crc:4967
  3: F103-Mecrisp     20500b  crc:A4D0
  4: F103-SerPlus      7052b  crc:7DD0
  5: L052-Blink         568b  crc:311E
  6: L052-Mecrisp     20500b  crc:00AC
Use '!u <n>' to upload a specific one.
!u 2
....................................
............^C`

What I am missing???
regards,

tve, 2017-06-10:
I'm not sure what you're trying to do... In the first step you flashed Mecrisp with USB driver onto your blue pill using st-linkv2. The you can connect to it with folie, which you did. But don't use !u, which would overwrite what you just flashed! You should hit ctrl-C to reset and you should then see the Mecrisp welcome line and an `ok.` prompt.
(Sorry, I'm not super familiar with suf and jcw is offline for a bunch of days.)

mvcorrea, 2017-06-11:
tve, thanks for your reply,

I just did everything again. It just flashed as normal, it says connected but no forth prompt "**ok.**".
reset "^c" a few times.. "^d" quits as normal
Can you please just check the output, case I am missing something?

```
path: /FORTH/embello/explore/1608-forth/suf

osx% st-info --probe
Found 1 stlink programmers
 serial: 503f6c06507555522844113f
openocd: "\x50\x3f\x6c\x06\x50\x75\x55\x52\x28\x44\x11\x3f"
  flash: 65536 (pagesize: 1024)
   sram: 20480
 chipid: 0x0410
  descr: F1 Medium-density device

!! set boot0 to 1 on the blue pill...
osx% st-flash --reset write usb-common.bin 0x8000000
st-flash 1.3.1
2017-06-11T19:18:58 INFO src/common.c: Loading device parameters....
2017-06-11T19:18:58 INFO src/common.c: Device connected is: F1 Medium-density device, id 0x20036410
2017-06-11T19:18:58 INFO src/common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x10000 bytes (64 KiB) in pages of 1024 bytes
2017-06-11T19:18:58 INFO src/common.c: Attempting to write 48128 (0xbc00) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0800b800 erased
2017-06-11T19:19:00 INFO src/common.c: Finished erasing 47 pages of 1024 (0x400) bytes
2017-06-11T19:19:00 INFO src/common.c: Starting Flash write for VL/F0/F3 core id
2017-06-11T19:19:00 INFO src/flash_loader.c: Successfully loaded flash loader in sram
 46/47 pages written
2017-06-11T19:19:02 INFO src/common.c: Starting verification of write complete
2017-06-11T19:19:02 INFO src/common.c: Flash written and verified! jolly good!

!! returned boot0 to 0 and reset the board
!! looks ok... I am able to see the device
osx% ls -la /dev/cu* 
crw-rw-rw-  1 root  wheel   21,   1 Jun 10 05:27 /dev/cu.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   21,  25 Jun 11 19:19 /dev/cu.usbmodemD5222A31

!! downloading the client
osx% wget https://github.com/jeelabs/folie/releases/download/v2.12/folie-macos64.gz
--2017-06-11 19:21:24--  https://github.com/jeelabs/folie/releases/download/v2.12/folie-macos64.gz
Resolving github.com... 192.30.253.112, 192.30.253.113
Connecting to github.com|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github-production-release-asset-2e65be.s3.amazonaws.com/55057795/d4d3bd82-3968-11e7-975e-dcbefa4b1431?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20170611%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20170611T182123Z&X-Amz-Expires=300&X-Amz-Signature=cd2cbc588f2867d8825844c70cb488af84b90d1d76e15e391908a926383a55d4&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dfolie-macos64.gz&response-content-type=application%2Foctet-stream [following]
--2017-06-11 19:21:25--  https://github-production-release-asset-2e65be.s3.amazonaws.com/55057795/d4d3bd82-3968-11e7-975e-dcbefa4b1431?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20170611%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20170611T182123Z&X-Amz-Expires=300&X-Amz-Signature=cd2cbc588f2867d8825844c70cb488af84b90d1d76e15e391908a926383a55d4&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3Dfolie-macos64.gz&response-content-type=application%2Foctet-stream
Resolving github-production-release-asset-2e65be.s3.amazonaws.com... 54.231.48.176
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com|54.231.48.176|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2237363 (2.1M) [application/octet-stream]
Saving to: ‘folie-macos64.gz’

folie-macos64.gz                  100%[=============================================================>]   2.13M  1.72MB/s    in 1.2s

2017-06-11 19:21:26 (1.72 MB/s) - ‘folie-macos64.gz’ saved [2237363/2237363]

osx% gunzip folie-macos64.gz

osx% ls
Makefile           f-common.fs        f-olip103.fs       usb-common.bin     usb-maplemini.bin  usb-olip103.bin
README.md          f-hotcbo.fs        folie-macos64      usb-common.hex     usb-maplemini.hex  usb-olip103.hex
dump.fs            f-maplemini.fs     fz-port103z.fs     usb-hotcbo.bin     usb-olimexino.bin  usb.fs
ex/                f-olimexino.fs     hal-stm32f1.fs     usb-hotcbo.hex     usb-olimexino.hex

osx% chmod u+x folie-macos64

osx% ./folie-macos64
Folie v2.12
Select the serial port:
  1: /dev/cu.Bluetooth-Incoming-Port
  2: /dev/cu.usbmodemD5222A31
2
Enter '!help' for additional help, or ctrl-d to quit.
Serial port busy
[connected to /dev/cu.usbmodemD5222A31]
!reset
!reset
3 2 + .  	<- doesnt work (unable to see the "ok" prompt)
```

mvcorrea, 2017-06-11:
feedback...
when trying to connect directly I got the ok prompt..
yah I get a "\n\r" problem in the output but at least I got some output.
I will try the previous version of folie

```
mvcorrea@osx% sudo cu -s 115200 -l /dev/cu.usbmodemD5222A31                                                                           
Password:
Connected.
  ok.
       ok.
          3 2 + . 5  ok.
                          ok.
                               ok.
                                    ok.
```

tve, 2017-06-11:
Something is not right... I don't think you should see a `usbmodem` device. I have the feeling you're seeing your st-link USB device, not your F103's, or something like that. I'm not entirely sure how it looks on MacOS (jcw's blog posts should show that). On linux it comes in as `/dev/serial/by-id/usb-JeeLabs_SerPlus_313B6B69-if00` (with your chip's hex ID) which is a symlink to `/dev/ttyACM0`. You need to connect to that device with folie, not the st-link you originally used (you also need to plug your blue pill directly into USB).

mvcorrea, 2017-06-11:
When I returned boot0 to 0 I removed the stlink an connected directly to the bluepill usb port