jeelabs/embello #83: UART IRQ hangs - is there anything to

movingfox, 2017-06-10:
I encountered a strange problem while using mecrisp-stellaris with Forth libraries slightly modified from embello: UART IRQ would hang after receiving some bytes. The following is my code running on STM32F429 Discovery, I believe you could repeat it on other STM32 boards by appending similar code.

I loopback PC10 and PC11 together to run the following IRQ experiment. Can't figure out the problem because it hangs without any clue.

```
\ interrupt-based USART3 with input ring buffer
\ needs ring.fs,cond.fs,stm32f4/io.h

$40020800 constant GPIOC  

$40004800 constant USART3
   USART3 $00 + constant USART3-SR
   USART3 $04 + constant USART3-DR
   USART3 $08 + constant USART3-BRR
   USART3 $0C + constant USART3-CR1
\  USART3 $10 + constant USART3-CR2
\  USART3 $14 + constant USART3-CR3
\  USART3 $18 + constant USART3-GPTR

: usart3-baud!  ( baud -- )
    clock-hz @ swap /  2 rshift USART3-BRR ! ;  \ APB1 clock is equal to 1/4 system clock

: usart3-init ( -- )  
    2 bit RCC-AHB1ENR bis!                  \ IO port C clock enabled
    OMODE-AF-PP OMODE-SLOW + PC10 io-mode!  \ PC10 -> Alternate function mode
    OMODE-AF-PP OMODE-SLOW + PC11 io-mode!  \ PC11 -> Alternate function mode
    $0700 GPIO.AFRH GPIOC + bis!            \ PC10 -> Alternate function: %0111: AF7 (USART3 TX) 
    $7000 GPIO.AFRH GPIOC + bis!            \ PC11 -> Alternate function: %0111: AF7 (USART3 RX) 

    18 bit RCC-APB1ENR bis!     \ Enable clock for USART3
     2 bit USART3-CR1 bis!      \ Receiver enable
     3 bit USART3-CR1 bis!      \ Transmitter enable
    13 bit USART3-CR1 bis!      \ USART enable
    2400 usart3-baud!           \ 2400 baud
;

: usart3Txe? ( -- f )
    7 bit USART3-SR bit@ inline ;   \ TXE

: usart3Rxne? ( -- f )
    5 bit USART3-SR bit@ inline ;   \ RXNE

\ transmit a byte to USART
: >usart3 ( byte -- )
    begin pause usart3Txe? until    \ wait until transmit buffer is empty
    USART3-DR h!                    \ send byte
;

\ read a byte from USART
: usart3> (  --  byte )
    begin pause usart3Rxne? until   \ wait until receive buffer is not empty
    USART3-DR h@                    \ fetch data
;

$E000E104 constant NVIC-EN1R        \ IRQ 32 to 63 Set Enable Register

128 4 + buffer: usart3-ring

: usart3-irq-handler ( -- )         \ handle the USART receive interrupt
  USART3-DR @   \ will drop input when there is no room left
  usart3-ring dup ring? if >ring else 2drop then
;

: usart3-irq-init ( -- )    \ initialise the USART3 using a receive ring buffer
  usart3-ring 128 init-ring
  ['] usart3-irq-handler irq-usart3 !
  7 bit NVIC-EN1R !         \ enable USART3 interrupt 39
  5 bit USART3-CR1 bis!     \ set RXNEIE
;

: usart3-irq-key? ( -- f )  \ input check for interrupt-driven ring buffer
  pause usart3-ring ring# 0<> ;
: usart3-irq-key ( -- c )   \ input read from interrupt-driven ring buffer
  begin usart3-irq-key? until  usart3-ring ring> ;

\ : init ( -- )
\   [ifdef] init init [then] \ run previous init if it exists
\   usart3-irq-init
\   ['] usart3-irq-key? hook-key?  !
\   ['] usart3-irq-key  hook-key   !
\   ;

task: sender-task

: sender&
    sender-task activate
    begin
        1 ms
        [CHAR] A >usart3
    again
;

task: catcher-task

: catcher&
    catcher-task activate  begin  usart3-irq-key emit  again
;

task: blinker-task

: blinker&
    OMODE-PP PG13 io-mode!
    blinker-task activate
    begin
        500 ms
        PG13 iox!
    again
;

: launch
    usart3-init
    usart3-irq-init
    lowpower&
    blinker&
    catcher&
    sender& ;
```

tve, 2017-06-10:
One issue I see is that in the sender task you're trying to send a char every 1ms and at 2400 baud each char  takes 4.16ms. You can either remove the `1 ms` or make it longer if you wanted to space chars out. I would try an experiment with a 100ms delay first and see whether that runs for a while. If it does, there's a problem with the buffer full situation, else there's a problem with the general tx/rx code. When it stops, does the TX continue?

I'm not familiar with the F4 series' usart but I notice that you use `h!` and `h@` to access the DR register, is that appropriate? Shouldn't it be `!` and `@`?

Not sure what `lowpower&` starts...

(FYI, jcw is offline for a few days)

movingfox, 2017-06-10:
Hi @tve , thanks for your kindly suggestions.

Actually I noticed this problem first when I tried to build the communication with an external MCU. Then I setup this loopback test to investigate the bug. This problem is very strange because it hangs even that I drop what it receives in the IRQ handler. Then I don't know if there's anything I should do to end the IRQ handler gracefully.

I've tried to make the delay longer and remove the 'h' prefix (the code is copied from mecrisp / embello as well)

`lowpower&` could be removed as you wish.

pelrun, 2017-08-18:
This is most likely because the UART error states aren't correctly handled. The big one is that enabling RXNEIE *also* enables the ORE overrun interrupt. If it's not explicitly handled and cleared, then when it occurs it masks RXNE and you don't see any more characters.

gerey73, 2018-03-06:
Have you been able to solve this problem? I have a similar problem. Until I understood, that uart-irq and multitask together not want work. Although separately work normally!

gerey73, 2018-06-21:
in theory, in your example there is no word _**multitask**_