jeelabs/embello #88: OLED  SSD1306  and Blue Pill

enriquepinedo, 2017-10-04:
Hi to everybody,   Pls., I m asking for advise

how and where to find the files  to setup a small OLED display with a Blue Pill generic STM32F103

I am totally confused with all the information and files and renames of files (oled etc) on your github
I only want to try the Oled interface or driver from Mecrisp (which is running fine on my STM32f board)

My OLED has  4 PINS  /    GND VDD    SCK  SDA 
I have triyed some of your files without any success (the one with the graphic output, the I2c include etc)

Can you help me to find the links to 2 or 3 files that will install (including the includes links)
and init the display and  write  "hello world"  to the OLED ?   and also where is the pin config of these
drivers ?

Kind regards
Enrique

(btw many of your examples on the blog the links are broken)

jcw, 2017-10-04:
Does http://embello.jeelabs.org/flib/ help?

enriquepinedo, 2017-10-04:
I found this article :  http://embello.jeelabs.org/flib/i2c.html
and I tested this :
https://github.com/jeelabs/embello/blob/master/explore/1608-forth/flib/any/i2c-bb.fs

I get all lines  "--"  empty, this means STM32 will not recognize any OLED at I2C,  I added
at PB6 and PB7  2  10 kohm  pull-up to 3.3 V  as somebody suggested,  no success...

jcw, 2017-10-04:
Sounds good to me. What are the commands you used to test? SDA/SCL swapped? Power and ground connected to the OLED? Is the OLED set for I2C, not SPI?

enriquepinedo, 2017-10-04:
Dear Jan, first of all  thank you for your answers !  
and also the wonderful projects presented in your site and forth language examples, I like everything very much !
It is a kind of spooky thing this I2c.... I finally got 1 or 2twice the   message  the Oled was detected at 3C

I am at  I2C as far as I understand, I could detect it 1 time atlast...

I have found your article 1708a  where it shows the same OLED 4 wires I am using, and it says 
"i2c-init i2c".     
my error was to make only  i2c.  without  the   i2c-init
*******************************************************************

But after that  I could not  move forward , I did never got your Logo on the OLED (following
the example on your site) , it did happend nothing....
only  2 times I  detected   the periferal at $3C  (without any pullup)

Now , I can´t again ....   it  appears   all    "--"  .

I did also pullups of  1k  , and 10 k,  nothing changes....
I changed also the PB6 & 7   for  4 & 5   no success....

I triyed also to flash this code, so I do not need to reload it again, 
But I ve got   an error after restart   (interrupt 00003) ...

I will contact you as soon I see where the error can be, now I have this good example to follow
and the  big-bang i2c file I was not shure where to look for.

Thanks again.

enriquepinedo, 2017-10-04:
I added  i2c-half   and now EVERYTHING WORKS !
it was a timing problem.

thanks again  will keep studying Mecrisp

jcw, 2017-10-04:
Good to hear you're making progress. The timing can be tricky. Try adding the following definition _before_ including the `i2c-bb.fs` code:

    9 constant I2C.DELAY

It inserts delay loops to slow down I2C comms (I use 9, sometimes 50 for really slow chips).

jcw, 2017-10-06:
In case you're interested - I've been working on a new documentation site, see <https://docs.jeelabs.org> - my first goal is merely to start organising the information out there (which, as you pointed out, is strewn all over the place), leaving out whatever is no longer applicable. It's not going to be done overnight, but if you have tips on things I should add / mention / fix, please let me know.

enriquepinedo, 2017-10-06:
Thank you Jan I appreciate very much. Btw.  Nice design !

There is really a sea of information  !  You have done a terrific job !  I like to follow information
in the form of a book with chapters, and not  in a form of  individual pages.  I recovered my 
taste for Forth (in love with Forth since 35 years btw)  thanks  Hakcaday-Mecrisp and YOUR site.
Thumbs Up for you and everybody involved in this project !

closed 2017-10-06 by enriquepinedo