jeelabs/embello #90: Confusion with Maple Mini Baite (ex red pill... )

enriquepinedo, 2017-10-14:
Hello friends, 

Here again asking for your kind advise and experience....
I have 2 boards STM32 one  Blue Pill with the jumpers for programming runs excellent
with Mecrisp embello, thanks everybody for such a great job.

Now I have a 2nd board (cheap clone of Maple Mini) from Baite 
http://wiki.stm32duino.com/index.php?title=Maple_Mini

I can programm it from Arduino IDE , I installed the drivers , ok recognized no problem

BUT   , I can not make it communicate through a USB to TTL board(ftdi f.ex)  , to upload Mecrisp.
I did everything possible, I triyed with Folie -r,  it keeps  sending ............    because there is
no communication, I Triyed "STMFlashLoader Demo"  (with this one I uploaded Mecrisp binay
to the other Blue Pill with the Boot jumper BOOT0=1  the other at 0.  etc etc ....
There is no way, I followed many dif. sites explaining how to programm it , and could not do it.

I am also a bit confused, because Maple Mini has many (I think 3 different UARTS (I triyed all of them))
1) which PINS are RFX and TX for programming ?  
2) Does this happend becasue of the bootloader ?
3) Can I use a USBASP programmer as ST-link programmer ?  ( I triyed also to use STM32-STLINK utility
with my FTDI chip, but it keeps saying there is no STlink attached.... could not move forward)
4) Which are the procedures to make a Mapple Mini enter into programming mode, for example 
If I try to use Resett+ But32 switches pressed, and then I free first reset while but32 is pressed, some
sites say this will force into programming mode, I can do that but I loose any USB connections because
the USB link (on board USB connector) will not work (my COM10) is lost.....

This is all really very confusing, so I must give up and ask the experts in this forum for advise.

Thank you in advance for your help !
Regards
Enrique

enriquepinedo, 2017-10-15:
The problem of the above  was resolved

I could programm this Baite Maple Mini.... 
  (but embello is not working) , I was missing to place "but" pin (on my poor silkscreen mask) it is the 3rd pin after vcc and gnd from below
to VCC. Apparently pressing the BUT button was not enough after I did this jumper I could programm the board from a FTDI chip   using STm32loader.py   and following  instructions of  .
http://stm8sdiscovery.blogspot.com.br/2017/05/mecrisp-stellaris-on-maple-mini-clone.html

Now on yours Blogs  says  PB9 must be set to VCC in order to make USB work....
Somebody can tell me which one is the PB9 pin on this board ?
http://wiki.stm32duino.com/index.php?ti ... o_Logo.png

I Think my Board has no PB9... can somebody confirm this ?

Thanks again
Enrique

jcw, 2017-10-15:
Glad you worked it out.

PB9: this is an internal signal which enables USB. You must use the `usb-maplemini` build of the USB-console in [this folder](https://github.com/jeelabs/embello/tree/master/explore/1608-forth/suf). IOW, no need to connect anything, but make sure you flash `usb-maplemini.bin` or `usb-maplemini.hex` to your board. Other images won't work. The logic is defined in [f-maplemini.fs](https://github.com/jeelabs/embello/blob/master/explore/1608-forth/suf/f-maplemini.fs).

enriquepinedo, 2017-10-15:
Dear Jan !
Thanks again for your help.
Shure, I did burn the usb-maplemini.bin,  after that I did not see anything in Teraterm 115kb working at this com.
(Trough an FTDI chip connected on UART1 tx1 rx1 ). I also take all jumpers of boot1 and 0 from the chip.
If I connect directly the USB on board of the maple mini , Windows7 says there is no com present.
I could (before burning mecrisp) install this Maple mini as device driver on Win7 and using Arduino Ide
I could programm a blink sketch without any problems,. And Windows recognized the USB Maple and its
COM.  After installing Mecrisp  usb-maplemini.bin   nothing happends.  
Can this be a bootloader problem ?  
I followed this explanation  of Mr Foo's blog  
http://stm8sdiscovery.blogspot.com.br/2017/05/mecrisp-stellaris-on-maple-mini-clone.html

Best regards
Enrique

enriquepinedo, 2017-10-18:
Dear Jean 

I did it work  but  flashing   "mecrisp-stellaris-stm32f103-ra"  
i could not  make it work with    usb-minimaple.bin  

I am using the USB-ttl device to communicate trough RX TX of uart1.

USB-mapple-mini.bin   does not work for me, in 2 diff. PCs with Windows.  Of course
It can be  a problem of drivers too, that windows is not recognizing the USB of this baite-maple-clone blue pill. But for shure it is strange 2 diff. pcs....
(  the" normal blue pill",   works  well in USB the one with the 2 jumpers)...

any suggestion what to change or experiment... are welcome.

Best regards
Enrique

enriquepinedo, 2017-10-21:
Today I finally managged to install a usb-maple-mini.bin into the  "Baite clone"
It works perfect !  Thanks Jean ! 
I could do it trough folie -r  /  and then    !u  usb-maple-mini.bin   
(suddenly it stopped  doing  .....  , and started flashing giving the % of completion)

What was probably wrong and why I could not achieve in the first tries.... ?

I presume all the problems with this type of blue-pill or better Maple-mini clone
is in the confusion with the reset buttons , reset procedure, and the jumpers one
low (button)  and  the other high (but-pin 3 from buttom up) .
It is also very confusing that when once this maple-mini is programmed (from outside
of arduino-ide)  the only led that is on board, is dead... So you really never know
if the board is alive, burned from a mistake, or in what kind of state.

Probably I have done more than 20 temptatives with different kind of flashing softwares , ides, 
etc to get it working in USB mode (installing mecrisp-ttl rs232 was no problem as mentioned
above).

Once again thanks for this space, and thumbs up to  jeenode and embello !
Regards
Enrique

closed 2017-10-21 by enriquepinedo

jcw, 2017-10-21:
Thanks for following up. Yes, there's a lot of confusion around all this STM32 first-time uploading, and it differs across the many board types. Great to hear that you got it all working in the end!