jeelabs/embello #94: cornerstone not found

osterwood, 2018-02-28:
I'm having an issue and am a bit lost.  I got a STM32F103 running the other day, but now (repeating what I think are the same steps) I am seeing a error when loading board.fs

Output from sending `always.fs`

```
!send always.fs
1> always.fs 3: Erase block at  00005000  from Flash
1> always.fs 4: Erase block at  00005400  from Flash
Erase block at  00005800  from Flash
Erase block at  00005C00  from Flash
Erase block at  00006000  from Flash
Erase block at  00006400  from Flash
Erase block at  00006800  from Flash
Erase block at  00006C00  from Flash
Erase block at  00007000  from Flash
Finished. Reset ?Mecrisp-Stellaris RA 2.3.5 for STM32F103 by Matthias Koch
1> always.fs 25: Redefine eraseflash.  ok.
```

I reboot the MCU and send `board.fs`, and get the following:

```
!send board.fs
1> board.fs 3: Finished. Reset ?Mecrisp-Stellaris RA 2.3.5 for STM32F103 by Matthias Koch
1> board.fs 5: ( board start: ) 00005000  ok.
1> board.fs 28: Redefine init.  ok.
1> board.fs 55: cornerstone not found.
Send failed. 
```

I don't understand why `board.fs` starting writing to the 0x5000 address, that seems to be overwriting the definitions created by `always.fs`

When I put `( flash use: ) here hex.` at the end of `always.fs`, is prints `( flash use: ) 200004CC` and the end of the load and then `board.fs` results in 
```
Erase block at  20000800  from Flash
Erase block at  20000802  from Flash
Erase block at  20000804  from Flash
Erase block at  20000806  from Flash
Erase block at  20000808  from Flash
Erase block at  2000080A  from Flash
Erase block at  2000080C  from Flash
Erase block at  2000080E  from Flash
Erase block at  20000810  from Flash
```

And that continues for a long time until the system crashes.

Is there a "master" example of `always.fs` and `board.fs` that I should be using?  As I said, I loaded a set yesterday and got my system working, but can't seem to recreate that today.

My base image is `g6s-release.bin`, created from the `g6s-release.hex` in `explore/1608-forth/g6s`

jcw, 2018-02-28:
There may be several intermingled issues at play. I've seen the `Erase block at  20000800` message a few times, and have never been able to track it down - for some reason, the `compiletoflash` did not take, causing the `cornerstone` definition to end up in RAM iso flash. That then gets lost in the reset.

I think that the issue can be avoided by hitting return, to check that you get an `ok.` prompt before running `!s ...`, but it's definitely weird and not consistent across retries.

Hm - perhaps start from the latest `cores/f103-mecrisp-2.4.1.bin` file instead. If you let me know how that goes, maybe we can then figure it out and come up with a newer (and working) `g6s-release.bin`.

osterwood, 2018-03-01:
I tried the 2.4.1 image and got the same thing:

```
!send always.fs
1> always.fs 3: Erase block at  00005000  from Flash
1> always.fs 4: Erase block at  00005400  from Flash
Erase block at  00005800  from Flash
Erase block at  00005C00  from Flash
Erase block at  00006000  from Flash
Erase block at  00006400  from Flash
Erase block at  00006800  from Flash
Erase block at  00006C00  from Flash
Erase block at  00007000  from Flash
Finished. Reset ?Mecrisp-Stellaris RA 2.4.1 for STM32F103 by Matthias Koch
1> always.fs 24: ( flash use: ) 200004CC  ok.
1> always.fs 25: Redefine eraseflash.  ok.
  ok.
!send board.fs
1> board.fs 3: eraseflash 
Erase block at  20000800  from Flash
Erase block at  20000802  from Flash
Erase block at  20000804  from Flash
Erase block at  20000806  from Flash
Erase block at  20000808  from Flash
Erase block at  2000080A  from Flash
Erase block at  2000080C  from Flash
Erase block at  2000080E  from Flash
Erase block at  20000810  from Flash
Erase block at  20000812  from Flash
Erase block at  20000814  from Flash
Erase block at  20000816  from Flash
Erase block at  20000818  from Flash
Erase block at  2000081A  from Flash
Erase block at  2000081C  from Flash
```

I tried re-flashing, and it worked the second time

```
!send always.fs
1> always.fs 3: Finished. Reset ?Mecrisp-Stellaris RA 2.4.1 for STM32F103 by Matthias Koch
1> always.fs 24: ( flash use: ) 00005134  ok.
1> always.fs 25: Redefine eraseflash.  ok.
!send board.fs
1> board.fs 4: Finished. Reset ?Mecrisp-Stellaris RA 2.4.1 for STM32F103 by Matthias Koch
1> board.fs 6: ( board start: ) 00005400  ok.
7> hal.fs 5: Redefine chipid.  ok.
7> hal.fs 7: Redefine hwid.  ok.
7> hal.fs 9: Redefine flash-kb.  ok.
7> hal.fs 11: Redefine flash-pagesize.  ok.
1> board.fs 29: Redefine init.  ok.
1> board.fs 55: ( board end, size: ) 000076C4 8900  ok.
1> board.fs 59: 64 KB <g6s> 775BBF39 ram/flash: 19336 34816 free  ok.
```

I have no idea why `always.fs` started with `Erase Block` one time and not the other.  I didn't change the code.

osterwood, 2018-03-01:
By the way -- thanks for the quick reply and suggestions here.  Sorry that I'm not able to reproduce the problem.  I tried a third flash and had success again.

jcw, 2018-03-03:
I've added some dummy `cr` lines right after the commands which implicitly do a reset (which can eat up the next character if sent too quickly). It seems to fix the issue. Please re-open this issue if you still run into this.

closed 2018-03-03 by jcw