jeelabs/embello #97: modifying f103 example for f303re nucleo board

danbates2, 2019-01-24:
Hi, I'm trying to modify the f103 examples for the f303re discovery board.
I've got as far as modifying the makefile architecture flags, linker files, and GPIO definition in the blink.c file example but am getting compilation errors.
Any thoughts?
I'm getting:
........ uses VFP register arguments, blink.elf does not
and
...failed to merge target specific data of file...
several times.

This is the chip on the discovery board, it's an M4.

The ultimate aim is getting RFM69 radio working, at least in RX mode, for an energy monitoring project.

danbates2, 2019-01-24:
I modified the linker file, working now.
Radio next..
[stm32f303re.ld.zip](https://github.com/jeelabs/embello/files/2791869/stm32f303re.ld.zip)

closed 2019-01-24 by danbates2

JohnOH, 2019-01-24:
Please keep us in the loop on your progress with the radio.

danbates2, 2019-02-07:
Update:
I'm completely stuck after several attempts. I've translated the code to C.. there must be something I've missed but can't figure out what.

danbates2, 2019-02-08:
Hi again, I've made some progress now. Receiving something on the F303. I'm looking at the receiveDone code for the F103 now.. I've inserted prints to indicate where it gets to. I'm receiving packets, but don't know how to decode them. Seems to get this far:

`if (rfm69_readReg(REG_IRQFLAGS2) & IRQ2_PAYLOADREADY) {`

When I open RF12Demo on an arduino to check my transmitter I get:
```
Current configuration:
 A i1 g210 @ 433 MHz q1
OK 5 0 22 21 0 (-67) 
```

Below are my transmitter and receiver:
emonBase and an STM32 nucleo I'm dev'ing on for openenergymonitor.
[image]
[image]

Full receiveDone code below for reference if anyone fancies taking a look..

```
int rfm69_receive(void *ptr, int len) {
      if (mode != MODE_RECEIVE)
        rfm69_setMode(MODE_RECEIVE);
      else {
        static uint8_t lastFlag;
        if ((rfm69_readReg(REG_IRQFLAGS1) & IRQ1_RXREADY) != lastFlag) {
          lastFlag ^= IRQ1_RXREADY;

          if (lastFlag) { // flag just went from 0 to 1
            rssi = rfm69_readReg(REG_RSSIVALUE);
            printf("rssi:%d\n", rssi / 2);
            lna = (rfm69_readReg(REG_LNAVALUE) >> 3) & 0x7;
            printf("lna:%d\n", lna);
    #if RF69_SPI_BULK
            SPI_transfer8(REG_AFCMSB);
            afc = SPI_transfer8(0) << 8;
            afc |= SPI_transfer8(0);
            printf("afc:%d\n", afc);
    #else
            afc = rfm69_readReg(REG_AFCMSB) << 8;
            afc |= rfm69_readReg(REG_AFCLSB);
    #endif
          }
        }

        if (rfm69_readReg(REG_IRQFLAGS2) & IRQ2_PAYLOADREADY) {

    #if RF69_SPI_BULK
          SPI_transfer8(REG_FIFO);
          int count = SPI_transfer8(0);
          for (int i = 0; i < count; ++i) {
            uint8_t v = SPI_transfer8(0);
            printf("count:%d", count);
            if (i < len)
              ((uint8_t *)ptr)[i] = v;
          }
    #else
          int count = rfm69_readReg(REG_FIFO);
          for (int i = 0; i < count; ++i) {
            uint8_t v = rfm69_readReg(REG_FIFO);
            if (i < len)
              ((uint8_t *)ptr)[i] = v;
          }
    #endif

          // only accept packets intended for us, or broadcasts
          // ... or any packet if we're the special catch-all node
          uint8_t dest = *(uint8_t *)ptr;
          //  if ((dest & 0xC0) == parity) {
            uint8_t destId = dest & 0x3F;
            //if (destId == myId || destId == 0 || myId == 63)
              return count;
          //}
        }
      }
      return -1;
}
```

and in the main loop the count return is expected for this:

```
void radio_loop() {
    int len = rfm69_receive(rxBuf, sizeof(rxBuf));
    if (len >= 0) {
        printf("rf69 %04x%02x%02x%02x%04x%02x%02x%02x ",
                rf_freq, rf_group, rssi, lna, afc,
                rxBuf[0], rxBuf[1], len - 2);
        for (int i = 2; i < len; ++i)
            printf("%02x", rxBuf[i]);
        const char* sep = afc < 0 ? "" : "+";
        if (verbose)
            printf("  (%g%s%d:%d)", rssi * 0.5, sep, afc, lna);
        putchar('\n');
    }
}
```
[image]

JohnOH, 2019-02-09:
Good progress.
The header documentation is [here](https://jeelabs.org/2011/06/09/rf12-packet-format-and-design/).
The decode of OK 5 0 22 21 0 (-67) is this:
OK - crc in received payload is good.
210 - not seen above but present in payload, stripped by the decoder
5 - the originating node number, 3 high order bits of 000 have significance - see above link
0 - first byte of payload
22 - second byte of payload
21 - third byte of payload
0 - last byte of payload
(-67) - received signal strength in dB.

JohnOH, 2019-02-09:
If you have integers stored in your payload the decode of OK 5 0 22 21 0 (-67) the higher bytes of the integer are stored later in the payload.
<low Byte><High Byte>
        22               21
would be a 16 bit integer of value 5398.

JohnOH, 2019-02-09:
Are you using interrupts for receive (I hope you are)?
If you are there are issues around setting the mode rfm69_setMode(MODE_RECEIVE);
It is possible to get stray interrupts as the radio passes through other states on its way to being in receive mode. Here is the [rather ugly code](https://github.com/jeelabs/jeelib/tree/RFxConsole-RSSI) I use:
```
uint8_t setMode (uint8_t mode) {	// TODO enhance return code
    uint8_t c = 0;
    if (mode >= MODE_FS) {
        uint8_t s = readReg(REG_DIOMAPPING1);// Save Interrupt triggers
         // Mask FS PllLock and appropriate target mode Interrupt
        if (mode == MODE_TRANSMITTER) writeReg(REG_DIOMAPPING1, DIO0_FS_UNDEF_TX);
        else writeReg(REG_DIOMAPPING1, DIO0_FS_UNDEF_RX);
        
        writeReg(REG_OPMODE, (MODE_FS /*| MODE_SEQUENCER_OFF*/));
        while (readReg(REG_IRQFLAGS1 & (IRQ1_PLL | IRQ1_MODEREADY)) == 0) {
            c++; if (c >= 127) break;
        }
        writeReg(REG_DIOMAPPING1, s);        // Restore Interrupt triggers
    }
	if (mode != MODE_FS) writeReg(REG_OPMODE, (mode | MODE_SEQUENCER_OFF));
        
    while ((readReg(REG_IRQFLAGS1) & IRQ1_MODEREADY) == 0) {
        c++; if (c >= 254) break;
    }
    return c;	// May need to beef this up since sometimes we don't appear to setmode correctly
}
```

JohnOH, 2019-02-09:
Please excuse my scatter gun approach:
myId == 63
The above and reference to parity refers to a later radio hardware based protocol. 
I think that emonBase uses the classic Jeelabs protocol as I do.

JohnOH, 2019-02-09:
It is also a good idea to clear the AFC after each receive:
```
		writeReg(REG_AFCFEI, (AFC_CLEAR));
```
The RFM69 has a habit of adjusting (AFC) it receive way out of where you want it if it sees noise on the edge of its listening frequency.

JohnOH, 2019-02-09:
@martynj3 would be concerned at the radio so close to the RasPi hardware. Much talk of introduced radio noise.

danbates2, 2019-02-09:
Thank you J, looks good. I'll take a look at this in the next couple of days and let you know how it goes.

danbates2, 2019-02-11:
Looking at you setmode code now.
REG_DIOMAPPING1 I could find from jeelib, but not MODE_FS, IRQ1_PLL, or MODE_SEQUENCER_OFF.
What are the reg addresses and values?

JohnOH, 2019-02-11:
They are in my branch [here](https://github.com/jeelabs/jeelib/blob/RFxConsole-RSSI/RF69.cpp). If you are not using interrupts I don't think the special handling in setMode is required.

danbates2, 2019-02-11:
I would like to use DIO0 to flag a payloadready signal. That's how I'm doing it with the lowpowerlab library and seems to work reliably.
Are you picking up conventional Jee packets with different group/syncaddress register settings in that branch?
I see in that DIO0 is mapped to RSSI, admittedly I'm confused.

How to I get the conventional Jeelib packets decoded?

Looking at the packet structure you linked, I need to get Len and extract Len number of bytes from the FIFO buffer, and then just print out the bytes from a buffer[].. right?

JohnOH, 2019-02-11:
Payload ready isn't suitable for classic Jeelib packets since the maximum packet size is 66 for classic Jeelib and the radio hardware buffer is only 64. The choice of 66 from RFM12B legacy has caused quite a lot of pain over the years. 
The RFxConsole branch picks up any group or all groups very cleanly. 
I use RSSI interrupt because I like to process the RF in very fine detail. The code has the option to interrupt on sync match which is the standard classic Jeelib approach as in the master branch. 

> Looking at the packet structure you linked, I need to get Len and extract Len number of bytes from the FIFO buffer, and then just print out the bytes from a buffer[].. right? Yes, that right, after crc check of course.

danbates2, 2019-02-11:
OK thanks,
The crc check looks at the last two bytes only correct?

JohnOH, 2019-02-12:
That's right.